isaac-sim/IsaacLab #2438: [Bug Report] Simulation Breaks With More Than 32 Robots

htchr, 2025-05-07:
### Describe the bug

I am trying to simulate many robots to generate training data for reinforcement learning.

I am able to simulate 32 robots smoothly.

When I attempt to simulate more, they begin to clip into the ground, jump, bend joints they weren't command to, fly off, and crash the program.

This is a video with 32 robots, with the `dt=120`:

https://github.com/user-attachments/assets/b8174c52-25c4-4d83-99d6-b8fea5ff3de9

This is a video with 32 robots with `dt=60`:

https://github.com/user-attachments/assets/9bc962f4-1939-4040-b917-ac9bed47f638

This is a video with 64 robots with `dt=120`:

https://github.com/user-attachments/assets/94d4bfff-9cd3-4fba-8cdf-2b14f583597a

When the program crashes I get the following error:

```
Error executing job with overrides: []
Traceback (most recent call last):
  File "/workspace/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/workspace/isaaclab/scripts/reinforcement_learning/rsl_rl/train.py", line 183, in main
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "/workspace/isaaclab/_isaac_sim/kit/python/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 260, in learn
    loss_dict = self.alg.update()
  File "/workspace/isaaclab/_isaac_sim/kit/python/lib/python3.10/site-packages/rsl_rl/algorithms/ppo.py", line 258, in update
    self.policy.act(obs_batch, masks=masks_batch, hidden_states=hid_states_batch[0])
  File "/workspace/isaaclab/_isaac_sim/kit/python/lib/python3.10/site-packages/rsl_rl/modules/actor_critic.py", line 122, in act
    return self.distribution.sample()
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/torch/distributions/normal.py", line 73, in sample
    return torch.normal(self.loc.expand(shape), self.scale.expand(shape))
RuntimeError: normal expects all elements of std >= 0.0

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2025-05-07 17:26:31 [179,743ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2025-05-07 17:26:31 [179,745ms] [Warning] [omni.graph.core.plugin] Could not find category 'animation' for removal
2025-05-07 17:26:31 [179,940ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-05-07 17:26:31 [179,950ms] [Warning] [carb.audio.context] 1 contexts were leaked
2025-05-07 17:26:31 [180,023ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
There was an error running python
```

### System Info

Describe the characteristic of your environment:

- Commit: 9be0de5f6196374fa793dc33978434542e7a40d3
- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: RTX 4090
- CUDA: 12.4
- GPU Driver: 550.120

### Additional context

I made [this](https://discordapp.com/channels/827959428476174346/1366529561101209611/1366529561101209611) post in the Isaac Lab Discord channel about this issue.

I am using mecanum wheels which are difficult to simulate correctly. 

I followed the advice [on the forums](https://forums.developer.nvidia.com/t/have-anyone-had-success-with-mecanum-wheels-on-isaac/220820/13) and manually spec'd the colliders of the rollers with spheres in the URDF file which greatly improved the simulation performance over a direct URDF export from Fusion 360 where the robot was modeled. 

Setting `dt=120`, `decimation=2`, and `solver_position_iteration_count=32` were also vital to achieving stable simulations of upto 32 robots.

I have followed many URDF import tutorials to ensure nothing is going wrong in that process.

I have followed the [simulation stability guide](https://docs.omniverse.nvidia.com/kit/docs/omni_physics/latest/dev_guide/guides/articulation_stability_guide.html) as best I can to address any known shortcomings. 

On discord, StrainFlow suggested setting the dt to 60, but that also seemed to de-stabilize the environments.

In the study group, it was suggested that it could be a computer resource limitation.

64 robots seems to occupy 80-90% of my 4090 GPU, but no more than 25% of my 64GB ram.

They also suggested to run headless to simulate more robots, but I'm concerned about them acting erratically without being able to visually confirm if things are working.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] I am able to simulate more robots for efficient data generation for reinforcement learning.

RandomOakForest, 2025-05-08:
Thank you for posting this. There may be a workaround, see comments of #673.

RandomOakForest, 2025-06-07:
I will close this issue for now. Please reopen it if you still observe the crash. Thanks.

closed 2025-06-07 by RandomOakForest