isaac-sim/IsaacLab #2440: [Bug Report] Joint Friction/Armature Randomization Fails with Error

asuzukii, 2025-05-07:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

When adding a joint friction/armature randomization event in a DirectRLEnv, the assignment fails due to a shape error.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

1. Add joint randomization, e.g. in anymalc_env_cfg:

```
@configclass
class EventCfg:
    """Configuration for randomization."""

    # existing events...
    add_joint_randomization = EventTerm(
        func=mdp.randomize_joint_parameters,
        mode="startup",
        params={
            "asset_cfg": SceneEntityCfg("robot", body_names=".*"),
            "friction_distribution_params": (0.0, 0.0),
            "armature_distribution_params": (0.0, 0.0),
        },
    )
```

### System Info

Describe the characteristic of your environment:

- Commit: 82cb320
- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: Quadro RTX 4000
- CUDA: 12.4
- GPU Driver: 550.120

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ]  Getting the assignment to properly work again

RandomOakForest, 2025-05-08:
Thank you for posting this and adding the PR. The team will review it. Thanks.

kellyguo11, 2025-06-25:
`randomize_joint_parameters` should be using `joint_names` instead of `body_names`. Please let us know if that works for you.

closed 2025-06-25 by kellyguo11