isaac-sim/IsaacLab #2441: [Bug Report] joint‑limit termination helpers raise *IndexError: too many indices* when joint_ids is set

GiulioRomualdi, 2025-05-07:
### Describe the bug

Calling `joint_pos_out_of_limit` or `joint_pos_out_of_manual_limit` with `asset_cfg.joint_ids` set to anything other than `None` raises

```
IndexError: too many indices for tensor of dimension 1
```

because the helpers reduce across joints (`torch.any(dim=1)`) **before** they slice with `asset_cfg.joint_ids`, leaving a 1‑D tensor that cannot be indexed with `[:, ids]`.

### Steps to reproduce

### System Info

* Commit: https://github.com/isaac-sim/IsaacLab/commit/82cb3206a443e566a67f64ebeb2aad810977c771 (current `main`)
* Isaac Sim Version: **4.5.0**
* OS: 22.04
* GPU: NVIDIA L40S
* CUDA: 12.4
* GPU Driver: 550.107.02

### Additional context

The bug breaks any workflow that relies on per‑joint termination masks
(e.g. training tasks that deactivate specific joints).

### Checklist

* [x] I have checked that there is no similar issue in the repo (**required**)
* [x] I have confirmed the error is in Isaac Lab helpers, not in Isaac Sim

### Acceptance Criteria

* [ ] `joint_pos_out_of_limit` works for `None`, `slice`, list and tensor joint selections.
* [ ] `joint_pos_out_of_manual_limit` shows the same correct behaviour.

GiulioRomualdi, 2025-05-07:
Fixed by https://github.com/isaac-sim/IsaacLab/pull/2442

RandomOakForest, 2025-05-08:
Thanks for posting this and adding the PR. The team will review it. Thanks!

closed 2025-07-18 by kellyguo11