isaac-sim/IsaacLab #2456: [Bug Report] Cannot modify network architecture with hydra if agent config is a configclass

ozhanozen, 2025-05-09:
### Describe the bug

I would like to modify the agent's neural network architecture using Hydra on the go, using command line arguments, e.g.: 

`isaaclab -p scripts/reinforcement_learning/rsl_rl/train.py --task my_task 'agent.policy.actor_hidden_dims=[512, 256, 128, 64]'`

However, if the agent's configuration is using the `configclass` (as in the `rsl_rl`), and the number of layers within the default config does not match the provided Hydra arguments, I encounter an error, specifically, I get the following exception:

```
File "/workspace/isaaclab/source/isaaclab/isaaclab/utils/dict-py", line 103,
in update_class_from_dict
raise ValueError(
ValueError: [Configl: Incorrect length under namespace: /policy/actor_hidden_dims. Expected: 5,
Received: 4.
```

Is there any workaround for this?

Thanks in advance!

### System Info

- Isaac Sim Version: 4.5
- OS: Ubuntu 22.04
- GPU: 4090 RTX
- CUDA: 12.2
- GPU Driver: 535.129.03

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] To be able to use Hydra to modify network structure

RandomOakForest, 2025-05-10:
Thanks for posting this. It's likely we have seen this before. Could you post a fuller error log? The team will review this. Thanks.

ozhanozen, 2025-05-10:
> Thanks for posting this. It's likely we have seen this before. Could you post a fuller error log? The team will review this. Thanks.

Thanks for your reply. A screenshot with the full error log is attached:

[image]

My understanding is that while updating the default config with Hydra arguments, the `update_class_from_dict()` function is called. This function considers that if the lengths of the Iterables do not match between the default config and Hydra arguments, this is an error, raising an exception. In other words, it is a deliberate design choice to raise this flag rather than a bug. However, this design choice results in the aforementioned limitation.

As a quick workaround, I tried to add the following exception for flat lists:

```
# If flat list, replace wholesale
if all(not isinstance(el, dict) for el in value):
   out_val = tuple(value) if isinstance(obj_mem, tuple) else value
   if isinstance(obj, dict):
      obj[key] = out_val
   else:
      setattr(obj, key, out_val)
   continue
```

right before this length check that cause the limitation:

https://github.com/isaac-sim/IsaacLab/blob/b1cd175f88d8315360095c96dfcfdab518f7778d/source/isaaclab/isaaclab/utils/dict.py#L102-L106

This seems to solve the issue/limitation. However, I am not sure if this would break something else, somewhere else. If you could provide feedback on this for a robust solution, I could create a PR accordingly.

RandomOakForest, 2025-05-13:
Thanks for submitting this. Would you like to submit a PR that we could review?

PR #2511 (2025-06-27): Modifies `update_class_from_dict()` to wholesale replace flat Iterables
# Description

Currently, `update_class_from_dict()` does not allow updating the configclass Iterable elements with Hydra, when the provided Iterable length differs from the default value. Such a feature is nevertheless needed when changing the network layer depth on the go, e.g., while using learning libraries that utilize configclass (see #2456 for details).

This PR modifies `update_class_from_dict()` such that if an element is a flat Iterable (e.g., flat list), it is replaced wholesale, without checking the lengths. Mo [... cut]
commit 394a1629 (2025-06-27): Modifies `update_class_from_dict()` to wholesale replace flat Iterables (#2511)

# Description

Currently, `update_class_from_dict()` does not allow updating the
configclass Iterable elements with Hydra, when the provided Iterable
length differs from the default value. Such a feature is nevertheless
needed when changing the network layer depth on the go, e.g., while
using learning libraries that utilize configclass (see #2456 for
details).

This PR modifies `update_class_from_dict()` such that if an element is a
flat Iterable (e.g., flat list), it is replaced wholesale, without
checking the le [... cut]