isaac-sim/IsaacLab #2458: [Question] pos_action_threshold and rot_action_threshold

pviswana-adi, 2025-05-09:
### Question

Hi, could you please explain the role of pos_action_threshold and rot_action_threshold variables defined in factory_env_cfg.py, which are using in _apply_action() function in factory_env.py?

RandomOakForest, 2025-05-10:
Thanks for posting this. We definitively need a better doc for this. It seems they just are scale factors in the target computations:
```
        # Interpret actions as target pos displacements and set pos target
        pos_actions = self.actions[:, 0:3] * self.pos_threshold

        # Interpret actions as target rot (axis-angle) displacements
        rot_actions = self.actions[:, 3:6]
        if self.cfg_task.unidirectional_rot:
            rot_actions[:, 2] = -(rot_actions[:, 2] + 1.0) * 0.5  # [-1, 0]
        rot_actions = rot_actions * self.rot_threshold
```

AntoineRichard, 2026-08-27:
@pviswana-adi thanks for pushing on this. They're scale factors, not gates: in Factory the normalized policy action is multiplied by `pos_action_threshold` [m] and `rot_action_threshold` [rad] to give the per-step target displacement, while `pos_action_bounds` clips the target relative to the fixed asset. FORGE differs — bounds map onto the workspace, thresholds clip per-step motion.

This is now written down in the `CtrlCfg` docstring and `ForgeEnv._apply_action` via https://github.com/isaac-sim/IsaacLab/pull/6158. Closing as answered; please reopen if anything there is still unclear.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard