isaac-sim/IsaacLab #2462: [Proposal] Request for adding distributed multi-GPU learning support for SB3 framework

jingyang-huang, 2025-05-10:
### Proposal

skrl, rl_games and rsl_rl all support multiple GPUs with command `--distributed` , when would it be available for sb3

### Motivation

SB3 is a extensive framework, I use it to train my model on the server which equips multi 3090 GPUs, but unfortunately I found that there`s no support for `--distributed` in SB3, I could accelerate my training with multiple GPUs.

### Alternatives

None

### Additional context

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] `--distributed` is available for sb3 and could train using multiple GPUs

RandomOakForest, 2025-05-10:
Thanks for posting this proposal. We are continuously working on improving the performance of Isaac Lab. The team will review this.

Mayankm96, 2025-05-12:
This unfortunately is not up to us. Last time I checked SB3 does not support distributed training. However, if this feature is supported now, our team can take a look into adding the support for it.

jingyang-huang, 2025-05-12:
> This unfortunately is not up to us. Last time I checked SB3 does not support distributed training. However, if this feature is supported now, our team can take a look into adding the support for it.

Thank you Mayankm, I think this would not be available in a few month, so I would switch to rsl_rl to continue my project

closed 2025-05-12 by jingyang-huang