isaac-sim/IsaacLab #2463: [Bug Report] Create IsaacLab container failed

KevinGeng3, 2025-05-10:
### Describe the bug

When I followed the instructions in https://isaac-sim.github.io/IsaacLab/main/source/how-to/cloudxr_teleoperation.html and used docker compose to run Isaac Lab and CloudXR runtime containers together:
`./docker/container.py start \
    --files docker-compose.cloudxr-runtime.patch.yaml \
    --env-file .env.cloudxr-runtime`
an error occurred:
```
[INFO] Using container profile: base
[INFO] X11 Forwarding is configured as '1' in '.container.cfg'.
	To disable X11 forwarding, set 'X11_FORWARDING_ENABLED=0' in '.container.cfg'.
[INFO] Building the docker image and starting the container 'isaac-lab-base' in the background...

Compose can now delegate builds to bake for better performance.
 To do so, set COMPOSE_BAKE=true.
[+] Building 2.8s (18/18) FINISHED                                                             docker:desktop-linux
 => [isaac-lab-base internal] load build definition from Dockerfile.base                                       0.0s
 => => transferring dockerfile: 4.20kB                                                                         0.0s
 => WARN: InvalidDefaultArgInFrom: Default value for ARG ${ISAACSIM_BASE_IMAGE_ARG}:${ISAACSIM_VERSION_ARG} r  0.0s
 => [isaac-lab-base internal] load metadata for nvcr.io/nvidia/isaac-sim:4.5.0                                 2.4s
 => [isaac-lab-base internal] load .dockerignore                                                               0.0s
 => => transferring context: 437B                                                                              0.0s
 => [isaac-lab-base base  1/12] FROM nvcr.io/nvidia/isaac-sim:4.5.0@sha256:c2f47dc82a7714af08d3766efe80ac9d08  0.0s
 => [isaac-lab-base internal] load build context                                                               0.0s
 => => transferring context: 108.65kB                                                                          0.0s
 => CACHED [isaac-lab-base base  2/12] RUN --mount=type=cache,target=/var/cache/apt     apt-get update && apt  0.0s
 => CACHED [isaac-lab-base base  3/12] COPY ../ /workspace/isaaclab                                            0.0s
 => CACHED [isaac-lab-base base  4/12] RUN ln -sf /isaac-sim /workspace/isaaclab/_isaac_sim                    0.0s
 => CACHED [isaac-lab-base base  5/12] RUN /workspace/isaaclab/isaaclab.sh -p -m pip install toml              0.0s
 => CACHED [isaac-lab-base base  6/12] RUN --mount=type=cache,target=/var/cache/apt     /workspace/isaaclab/i  0.0s
 => CACHED [isaac-lab-base base  7/12] RUN mkdir -p /isaac-sim/kit/cache &&     mkdir -p /root/.cache/ov &&    0.0s
 => CACHED [isaac-lab-base base  8/12] RUN touch /bin/nvidia-smi &&     touch /bin/nvidia-debugdump &&     to  0.0s
 => CACHED [isaac-lab-base base  9/12] RUN --mount=type=cache,target=/root/.cache/pip     /workspace/isaaclab  0.0s
 => CACHED [isaac-lab-base base 10/12] RUN /workspace/isaaclab/isaaclab.sh -p -m pip uninstall -y quadprog     0.0s
 => CACHED [isaac-lab-base base 11/12] RUN echo "export ISAACLAB_PATH=/workspace/isaaclab" >> ${HOME}/.bashrc  0.0s
 => CACHED [isaac-lab-base base 12/12] WORKDIR /workspace/isaaclab                                             0.0s
 => [isaac-lab-base] exporting to image                                                                        0.1s
 => => exporting layers                                                                                        0.0s
 => => writing image sha256:5039793e1506b8c35b09fb18c0062a1700b5fa7ac73a0645df1745ed009169cb                   0.0s
 => => naming to docker.io/library/isaac-lab-base                                                              0.0s
 => [isaac-lab-base] resolving provenance for metadata file                                                    0.0s
[+] Building 1/1
 ✔ isaac-lab-base  Built                                                                                       0.0s 
Compose can now delegate builds to bake for better performance.
 To do so, set COMPOSE_BAKE=true.
[+] Building 925.5s (18/18) FINISHED                                                           docker:desktop-linux
 => [isaac-lab-base internal] load build definition from Dockerfile.base                                       0.0s
 => => transferring dockerfile: 4.20kB                                                                         0.0s
 => WARN: InvalidDefaultArgInFrom: Default value for ARG ${ISAACSIM_BASE_IMAGE_ARG}:${ISAACSIM_VERSION_ARG} r  0.0s
 => [isaac-lab-base internal] load metadata for nvcr.io/nvidia/isaac-sim:4.5.0                                 0.5s
 => [isaac-lab-base internal] load .dockerignore                                                               0.0s
 => => transferring context: 437B                                                                              0.0s
 => [isaac-lab-base base  1/12] FROM nvcr.io/nvidia/isaac-sim:4.5.0@sha256:c2f47dc82a7714af08d3766efe80ac9d08  0.0s
 => [isaac-lab-base internal] load build context                                                               0.0s
 => => transferring context: 108.65kB                                                                          0.0s
 => CACHED [isaac-lab-base base  2/12] RUN --mount=type=cache,target=/var/cache/apt     apt-get update && apt  0.0s
 => CACHED [isaac-lab-base base  3/12] COPY ../ /workspace/isaaclab                                            0.0s
 => CACHED [isaac-lab-base base  4/12] RUN ln -sf /isaac-sim /workspace/isaaclab/_isaac_sim                    0.0s
 => CACHED [isaac-lab-base base  5/12] RUN /workspace/isaaclab/isaaclab.sh -p -m pip install toml              0.0s
 => CACHED [isaac-lab-base base  6/12] RUN --mount=type=cache,target=/var/cache/apt     /workspace/isaaclab/i  0.0s
 => CACHED [isaac-lab-base base  7/12] RUN mkdir -p /isaac-sim/kit/cache &&     mkdir -p /root/.cache/ov &&    0.0s
 => CACHED [isaac-lab-base base  8/12] RUN touch /bin/nvidia-smi &&     touch /bin/nvidia-debugdump &&     to  0.0s
 => CACHED [isaac-lab-base base  9/12] RUN --mount=type=cache,target=/root/.cache/pip     /workspace/isaaclab  0.0s
 => CACHED [isaac-lab-base base 10/12] RUN /workspace/isaaclab/isaaclab.sh -p -m pip uninstall -y quadprog     0.0s
 => CACHED [isaac-lab-base base 11/12] RUN echo "export ISAACLAB_PATH=/workspace/isaaclab" >> ${HOME}/.bashrc  0.0s
 => CACHED [isaac-lab-base base 12/12] WORKDIR /workspace/isaaclab                                             0.0s
 => [isaac-lab-base] exporting to image                                                                        0.1s
 => => exporting layers                                                                                        0.0s
 => => writing image sha256:5039793e1506b8c35b09fb18c0062a1700b5fa7ac73a0645df1745ed009169cb                   0.0s
 => => naming to docker.io/library/isaac-lab-base                                                              0.0s
 => ERROR [isaac-lab-base] resolving provenance for metadata file                                            924.6s
------
 > [isaac-lab-base] resolving provenance for metadata file:
------
rpc error: code = Unavailable desc = error reading from server: unexpected EOF
```
Does anyone know how to solve this problem? Thank you in advance!

### System Info

OS: ubuntu22.04

Isaac Lab: v2.1.0

docker version:
```
Client: Docker Engine - Community
 Version:           28.1.1
 API version:       1.48 (downgraded from 1.49)
 Go version:        go1.23.8
 Git commit:        4eba377
 Built:             Fri Apr 18 09:52:10 2025
 OS/Arch:           linux/amd64
 Context:           desktop-linux

Server: Docker Desktop 4.40.0 (187762)
 Engine:
  Version:          28.0.4
  API version:      1.48 (minimum version 1.24)
  Go version:       go1.23.7
  Git commit:       6430e49
  Built:            Tue Mar 25 15:07:22 2025
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.7.26
  GitCommit:        753481ec61c7c8955a23d6ff7bc8e4daed455734
 runc:
  Version:          1.2.5
  GitCommit:        v1.2.5-0-g59923ef
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

RandomOakForest, 2025-05-10:
Thanks for posting this. It seems we will have to repro this issue. The team will review it. Thanks.

KevinGeng3, 2025-05-11:
I've solved this problem. It was caused by net proxy.

closed 2025-05-11 by KevinGeng3