isaac-sim/IsaacLab #2466: [Bug Report] [bug] Mobile Base yaw not updating xy translation

exaFLOPs26, 2025-05-10:
### Describe the bug

Mobile bases with no distinct wheels (e.g., ridgeback Franka) have a hierarchical joint setup: world -> dummy_joint_x -> dummy_joint_y -> mobile_base. So even if I rotate the yaw, xy is not affected. So it goes like this.

https://github.com/user-attachments/assets/555108f9-3973-48f6-9c68-efc633ac2450

### Steps to reproduce

1. First I change the x_joint and it went forward and backward direction.
2. As I rotate the yaw for around 90 degree x,y should have changed.
3. Last I change the y_joint that the direction should have been forward and backwards but still sideways, which means that the yaw rotation is not updating the xy translation. And it is obvice by the hierarchical joint structure.

### What I did first

Change it to yaw-x-y joint hierarchical structure. But now it rotates in a circle. This is obvice even I change the xy translation, again in the opposite way, xy translation is not updating yaw rotation. So the center of yaw rotation is fixed at the first time.

### What I am doing

I use simple linear algebra to calculate the xy translation with yaw rotation. But I can't  move and rotate at the same time. And this is a big bottleneck on Imitation Learning and RL for mobile ones. 

[image]

### System Info

Describe the characteristic of your environment:

- Isaac Sim Version: 4.5.0
- OS: Ubuntu 22.04
- GPU: Nvidia L40S

[image]

### Additional context

- To do Sim2Real Real2Sim is important. How can we tune the simulation to make it stablized? Tried gain tuner, but at the end of the day, I did it manually.

- I remember a speech by Eric in office hour session, that what we need to do is not make the simulation same as the real world, but make it cover it. Are there any tips on this also?

[image]

exaFLOPs26, 2025-05-12:
Thanks! One more thing is that if I do ik absolute control for these kind of mobile manipulator(without distinct wheel like ridgeback_franka) it occurs a problem. 

I fixed the 'source/isaaclab/isaaclab/envs/mdp/task_space_actions.py' to make it work! If interested, I can share it.

RandomOakForest, 2025-05-12:
That would be great. Please add a PR for the team to review.

exaFLOPs26, 2025-05-13:
First I control the x joint and next I controled the Y joint. As I rotate 90 degree, wouldn't the x and y head the same direction?

I think my demo is quite overcomplicated, but the problem I am having is that the robot is not going forward toward the heading direction but just goint along the fixed world x axis even the yaw is rotated.

exaFLOPs26, 2025-05-13:
https://forums.developer.nvidia.com/t/moving-ridgebackfranka-robot-with-articulations-does-not-change-the-pose-of-the-robot/242112/8

exaFLOPs26, 2025-05-15:
I found out that others are having the same issue. How can we handle this?

exaFLOPs26, 2025-05-19:
@RandomOakForest Not to rush you, but this is still a bug to fix right?