isaac-sim/IsaacLab #2469: [Bug Report] Fatal Python error: Segmentation fault (core dumped) when running any Isaac Lab script

manuelecapece, 2025-05-11:
### Describe the bug

I installed Isaac Sim 4.5.0 and Isaac Lab following the recommended instructions (pip installation for Ubuntu 22.04). Isaac Sim works correctly, but Isaac Lab does not.

I noticed that Isaac Lab scripts only run when I specify the --headless argument. Otherwise, I get the following error: Segmentation fault (core dumped) ${python_exe} "$@"
I have tried multiple NVIDIA driver versions (535, 550, and 570), and I reinstalled everything several times, but the issue persists.
I’m attaching screenshots of my hardware and the terminal output when running the script to verify the installation: ./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py
(The same error occurs with other tutorial scripts as well.)

### Steps to reproduce

[image]

[image]

[image]

### System Info

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [4.5.0]
- OS: [Ubuntu 22.04]
- GPU: [RTX 3070]
- CPU: [AMD 7900X]
- CUDA: [12.2]
- GPU Driver: [535.230.02]

### Additional context

Previously, I was using a PC with an Intel i5-13600KF and everything worked fine. Now that I have replaced the CPU with an AMD Ryzen 9 7900X, this problem started to occur. I also completely reinstalled the operating system (Ubuntu 22.04), but that did not solve the issue.

RandomOakForest, 2025-05-12:
Thanks for posting this. Did the [verification step](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html#verifying-the-isaac-sim-installation) went well? Did you check the GLIB version? Have you tried the [binary install](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/binaries_installation.html#isaaclab-binaries-installation)?

manuelecapece, 2025-05-17:
I tried again this morning and now everything works correctly. The verification step and the GLIB version were fine. It will remain a mystery. Thanks for the support!

closed 2025-05-17 by manuelecapece