isaac-sim/IsaacLab #2478: [Question] ImplicitActuatorCfg effort_limit and effor_limit_sim, differences and values due to motor parameters

AndreaRossetto, 2025-05-13:
### Question

I have this motor parameters:

```

(Nominal Voltage)	                       V	       48
(Voltage Range)	                       V	       24~56
(Power)                                           W            336
(Nominal Torque)	                        N.M	       7.5
(Stall Torque)	                                N.M	        22
(Nominal Speed after Reduce)	RPM	110
(Max Speed after Reduce)	        RPM	320
(Nominal Current)	                        A	        7
(Stall Current)	                                A	        22
(Phase Resistance)	                        Ω	        0.67
(Phase Inductance)	                        mH	        0.45
(Speed Constant)	                        rpm/v	6.67
(Torque Constant)                           N.M/A      1
(Number of Pole Pairs)	                Pairs	21
(Gear Rate)	                                 /	        8:1

```
I'm using the ImplicitActuatorCfg, is the correct one to simulate a gear reduced brushless motor?
What is the difference between effort_limit_sim and effort_limit? 
The value for these two parameters is the Nominal Torque, the Stall Torque or something else?

Bardreamaster, 2025-05-14:
For ImplicitActuator, `effort_limit_sim` and `effort_limit` are the same to perform a PD control law to clip effort in model like [this](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.actuators.html#ideal-pd-actuator), neithor  Nominal Torque or Stall Torque. And the `effort_limit` will be removed in the future for ImplicitActuator.   `effort_limit_sim` is for setting the effort limit implicitly computed by simulator. For more information, try have a look on [changelog](https://github.com/isaac-sim/IsaacLab/commit/4c4377db3fe36319b6ee55f6a3fe77cf45f8fc4e) and [source code](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab/isaaclab/actuators/actuator_pd.py)

As I know, most brushless motors are DC motor, you might need [this](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.actuators.html#dc-motor-actuator). ImplicitActuator should not be used directly , which is informed in [docs](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.actuators.html#implicit-actuator).

Nate711, 2025-05-14:
@Bardreamaster The docs say not to use ImplicitActuator but the explicit actuator is so unstable with medium to large damping values that even the example models use implicit actuators: https://github.com/isaac-sim/IsaacLab/blob/fe6ee746ab606469baeafaa1aec9ecad1173e29e/source/isaaclab_assets/isaaclab_assets/robots/unitree.py#L217. Given that it's impractical to use the explicit actuators because of the instability, what's your recommendation?

RandomOakForest, 2025-06-11:
Thank you for following up. I'll move this to our _Discussions_ section for further follow-ups.