isaac-sim/IsaacLab #2481: [Bug Report] Bug title

SalmaZakaria, 2025-05-13:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

A clear and concise description of what the bug is.

### Steps to reproduce

I am trying to train the stacking task and it gives unsupported value type error, although the lifting task works properly.

```
Traceback (most recent call last):
  File "/home/salma_hassan/IsaacLab-SO100/Isaac-Lab-so100/scripts/rl_games/train.py", line 182, in <module>
    main()
  File "/home/salma_hassan/IsaacLab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 81, in wrapper
    env_cfg, agent_cfg = register_task_to_hydra(task_name, agent_cfg_entry_point)
  File "/home/salma_hassan/IsaacLab/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 59, in register_task_to_hydra
    ConfigStore.instance().store(name=task_name, node=cfg_dict)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/hydra/core/config_store.py", line 85, in store
    cfg = OmegaConf.structured(node)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/omegaconf.py", line 125, in structured
    return OmegaConf.create(obj, parent, flags)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/omegaconf.py", line 178, in create
    return OmegaConf._create_impl(
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/omegaconf.py", line 900, in _create_impl
    format_and_raise(node=None, key=None, value=None, msg=str(e), cause=e)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/_utils.py", line 819, in format_and_raise
    _raise(ex, cause)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/_utils.py", line 797, in _raise
    raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/omegaconf.py", line 861, in _create_impl
    return DictConfig(
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 111, in __init__
    format_and_raise(node=None, key=key, value=None, cause=ex, msg=str(ex))
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/_utils.py", line 819, in format_and_raise
    _raise(ex, cause)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/_utils.py", line 797, in _raise
    raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 109, in __init__
    self._set_value(content, flags=flags)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 647, in _set_value
    raise e
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 644, in _set_value
    self._set_value_impl(value, flags)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 690, in _set_value_impl
    self.__setitem__(k, v)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 314, in __setitem__
    self._format_and_raise(key=key, value=value, cause=e)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/base.py", line 231, in _format_and_raise
    format_and_raise(
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/_utils.py", line 819, in format_and_raise
    _raise(ex, cause)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/_utils.py", line 797, in _raise
    raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 308, in __setitem__
    self.__set_impl(key=key, value=value)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/dictconfig.py", line 318, in __set_impl
    self._set_item_impl(key, value)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/basecontainer.py", line 620, in _set_item_impl
    self._wrap_value_and_set(key, value, target_type_hint)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/basecontainer.py", line 636, in _wrap_value_and_set
    self._format_and_raise(key=key, value=val, cause=e)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/base.py", line 231, in _format_and_raise
    format_and_raise(
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/_utils.py", line 819, in format_and_raise
    _raise(ex, cause)
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/_utils.py", line 797, in _raise
    raise ex.with_traceback(sys.exc_info()[2])  # set env var OC_CAUSE=1 for full trace
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/basecontainer.py", line 628, in _wrap_value_and_set
    wrapped = _maybe_wrap(
[... 262 lines cut ...]
  File "/home/salma_hassan/.conda/envs/isaacLab/lib/python3.10/site-packages/omegaconf/nodes.py", line 154, in _validate_and_convert_impl
    raise UnsupportedValueType(
omegaconf.errors.UnsupportedValueType: Value 'slice' is not a supported primitive type
    full_key: env.events.randomize_cube_positions.params.asset_cfgs[0].joint_ids
    object_type=dict
```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:
- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. RTX 2060 Super]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

### Checklist

- [Yes] I have checked that there is no similar issue in the repo (**required**)
- [Yes] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-05-19:
Thank you for posting this. Could you fill in the description items for your bug report for your system info? these are the following items in the template above. Please also insert `[Bug Report]` at the beginning of your title. Thanks.
```
Commit: [e.g. 8f3b9ca]
Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by cat ${ISAACSIM_PATH}/VERSION]
OS: [e.g. Ubuntu 20.04]
GPU: [e.g. RTX 2060 Super]
CUDA: [e.g. 11.4]
GPU Driver: [e.g. 470.82.01, this can be seen by using nvidia-smi command.]
```

SalmaZakaria, 2025-05-25:
@RandomOakForest 
Isaac Sim Version: 4.5.0.0
OS: Ubuntu 20.04
GPU: NVIDIA GeForce RTX 4090
CUDA:  11.2

LinghengMeng, 2025-05-26:
This problem is related to `replace_slices_with_strings()` and `replace_strings_with_slices()` in _[IsaacLab/source/isaaclab/isaaclab/utils/dict.py](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab/isaaclab/utils/dict.py)_. The reason that this happens specifically to Isaac-Stack-Cube-Franka-v0 is in _[IsaacLab/source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/stack/config/franka/stack_joint_pos_env_cfg.py](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/stack/config/franka/stack_joint_pos_env_cfg.py)_ the franka_stack_events.randomize_object_pose takes a **list** of asset_cfgs as input as follows
```
randomize_cube_positions = EventTerm(
        func=franka_stack_events.randomize_object_pose,
        mode="reset",
        params={
            "pose_range": {"x": (0.4, 0.6), "y": (-0.10, 0.10), "z": (0.0203, 0.0203), "yaw": (-1.0, 1, 0)},
            "min_separation": 0.1,
            "asset_cfgs": [SceneEntityCfg("cube_1"), SceneEntityCfg("cube_2"), SceneEntityCfg("cube_3")],
        },
    )
```
while `replace_slices_with_strings()` and `replace_strings_with_slices()` do not iterate into itself when taking a list as an input.

To fix this problem, revise these two functions as follows
```
def replace_slices_with_strings(data: dict) -> dict:
    """Replace slice objects with their string representations in a dictionary.

    Args:
        data: The dictionary to process.

    Returns:
        The dictionary with slice objects replaced by their string representations.
    """
    if isinstance(data, dict):
        return {k: replace_slices_with_strings(v) for k, v in data.items()}
    elif isinstance(data, list):
        return [replace_slices_with_strings(v) for v in data]
    elif isinstance(data, slice):
        return f"slice({data.start},{data.stop},{data.step})"
    else:
        return data

def replace_strings_with_slices(data: dict) -> dict:
    """Replace string representations of slices with slice objects in a dictionary.

    Args:
        data: The dictionary to process.

    Returns:
        The dictionary with string representations of slices replaced by slice objects.
    """
    if isinstance(data, dict):
        return {k: replace_strings_with_slices(v) for k, v in data.items()}
    if isinstance(data, list):
        return [replace_strings_with_slices(v) for v in data]
    elif isinstance(data, str) and data.startswith("slice("):
        return string_to_slice(data)
    else:
        return data
```
to deal with a list of dicts with slice.

RandomOakForest, 2025-06-11:
Thank you for following up. The team will review this.

commit 367925cb (2025-07-08): Allows slicing to be processed from lists (#2853)

# Description

replace_slices_with_strings() and replace_strings_with_slices() in
[IsaacLab/source/isaaclab/isaaclab/utils/dict.py](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab/isaaclab/utils/dict.py)
are changed to allow slices being process if a list of dicts is passed
in as input.

Cherry picks changes from
https://github.com/isaac-sim/IsaacLab/pull/2571 by @LinghengMeng

Fixes #2481 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https:// [... cut]