isaac-sim/IsaacLab #2483: [Question] Question title

LULU-LULULA, 2025-05-13:
Title: Unable to Run IsaacLab on Remote Server with RTX 5090 GPU

Description:

I successfully installed Isaac Sim and IsaacLab on my server following the official instructions. However, when I try to run ./isaaclab.sh -p scripts/tutorials/00_sim/create_empty.py remotely from my local machine using VSCode, I encounter several errors. Additionally, I installed isaacsim-webrtc-streaming-client-1.0.6-linux-x64.AppImage locally to observe simulations remotely, but this attempt also failed.

System Configuration:

Server GPU: NVIDIA GeForce RTX 5090
Operating System: Ubuntu 22.04.5 LTS (Jammy Jellyfish)
Kernel Version: 6.8.0-57-generic
Driver Version: 570.133.07
Python Environment: Miniconda3 with Python 3.10
Error Logs:
Key errors include:

GLFW initialization failed multiple times.
Failed to startup plugin carb.windowing-glfw.plugin.
AttributeError: 'NoneType' object has no attribute 'get_window_minimize_event_stream'.
IRAY render warning: GPU with CUDA compute capability 12.0 is unsupported.
PhysX plugin startup failure.
Actions Taken:

Verified SSH configuration and remote connection settings in VSCode.
Checked and updated graphics drivers and necessary libraries (GLFW, etc.).
Ensured correct Python environment and path settings.
Attempted to use WebRTC streaming client for remote simulation observation.
Suspected Issues:

Potential incompatibility with the RTX 5090 GPU.
Missing software components or incorrect configurations on the server.
Possible version mismatch between installed components.
Request:
I am seeking guidance on resolving these issues to successfully run IsaacLab remotely. Any insights on additional software requirements or configurations would be greatly appreciated.

## The following is my detailed error message：
2025-05-13 10:06:23 [1,158ms] [Error] [omni.kit.window.drop_support.drop_support] Cannot setup ExternalDragDrop without a default window
[1.185s] [ext: omni.kit.property.adapter.core-1.0.2] startup
[1.186s] [ext: omni.inspect-1.0.1] startup
[1.186s] [ext: omni.kit.property.adapter.usd-1.0.2] startup
[1.187s] [ext: omni.graph.core-2.181.8] startup
[1.188s] [ext: omni.index.libs-380600.1777.0] startup
[1.188s] [ext: omni.hydra.rtx.shadercache.vulkan-1.0.0] startup
[1.189s] [ext: omni.ujitso.processor.texture-1.0.0] startup
[1.189s] [ext: isaacsim.core.deprecation_manager-0.2.2] startup
[1.190s] [ext: omni.index-1.0.1] startup
[1.190s] [ext: omni.kvdb-106.5.7] startup
[1.191s] [ext: omni.ujitso.client-0.0.0] startup
[1.191s] [ext: isaacsim.core.version-2.0.2] startup
[1.192s] [ext: omni.volume-0.5.0] startup
[1.195s] [ext: omni.localcache-106.5.7] startup
[1.196s] [ext: omni.usdphysics-106.5.7] startup
[1.198s] [ext: isaacsim.storage.native-1.0.6] startup
[1.198s] [ext: omni.hydra.rtx-1.0.0] startup
2025-05-13 10:06:23 [1,175ms] [Warning] [omni.log] Source: omni.hydra was already registered.
[1.210s] [ext: omni.convexdecomposition-106.5.7] startup
[1.213s] [ext: omni.kit.widget.highlight_label-1.0.3] startup
[1.213s] [ext: omni.physx.foundation-106.5.7] startup
[1.214s] [ext: omni.kit.widget.searchable_combobox-1.0.6] startup
[1.215s] [ext: omni.kit.menu.create-1.0.17] startup
[1.216s] [ext: omni.kit.window.property-1.11.5] startup
[1.217s] [ext: omni.kit.property.adapter.fabric-1.0.3] startup
[1.217s] [ext: omni.kit.widget.settings-1.2.2] startup
[1.219s] [ext: omni.physx.cooking-106.5.7] startup
[1.223s] [ext: omni.kit.property.usd-4.2.16] startup
[1.228s] [ext: omni.kit.window.preferences-1.7.0] startup
[1.229s] [ext: omni.physx-106.5.7] startup
[1.237s] [ext: omni.kit.scripting-106.5.2] startup
[1.239s] [ext: isaacsim.core.throttling-2.0.2] startup
[1.239s] [ext: isaacsim.replicator.behavior-1.0.8] startup
[... 741 lines cut ...]
[3.059s] [ext: isaaclab_assets-0.2.1] startup
[3.162s] [ext: isaaclab_tasks-0.10.25] startup
[3.670s] [ext: omni.kit.menu.common-1.1.9] startup
[3.671s] [ext: isaaclab_rl-0.1.1] startup
[3.911s] [ext: isaaclab_mimic-1.0.3] startup
[3.912s] [ext: isaaclab.python-2.0.2] startup
[3.914s] Simulation App Starting
2025-05-13 10:10:24 [4,225ms] [Error] [omni.kit.window.drop_support.drop_support] Cannot setup ExternalDragDrop without a default window
2025-05-13 10:10:24 [4,226ms] [Error] [omni.kit.window.drop_support.drop_support] Cannot setup ExternalDragDrop without a default window
[4.265s] app ready
2025-05-13 10:10:24 [4,246ms] [Error] [omni.kit.menu.utils.app_menu] Hotkeys cannot be setup without a default window
2025-05-13 10:10:24 [4,301ms] [Warning] [rtx.neuraylib.plugin] [IRAY:RENDER]   1.1   IRAY   rend warn : GPU 1 (NVIDIA GeForce RTX 5090) with CUDA compute capability 12.0 is unsupported by this version of iray photoreal. Please look for an update with your software provider.
2025-05-13 10:10:24 [4,305ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 7508933632
2025-05-13 10:10:24 [4,305ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid false, within: false
2025-05-13 10:10:24 [4,305ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : decrement: 167690, decrement size: 7433845248
2025-05-13 10:10:24 [4,305ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : New limit 9574251 (slope: 447, intercept: 13179904)
2025-05-13 10:10:24 [4,305ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit buffer size 4287216384
2025-05-13 10:10:24 [4,305ms] [Warning] [rtx.scenedb.plugin] SceneDbContext : TLAS limit : valid true, within: true
2025-05-13 10:10:24 [4,526ms] [Warning] [omni.usd-abi.plugin] No setting was found for '/rtx-defaults-transient/meshlights/forceDisable'
[6.354s] Simulation App Startup Complete
[6.404s] [ext: omni.physx.fabric-106.5.7] startup

(zenity:2902020): Gtk-WARNING **: 18:12:27.035: cannot open display: 

(zenity:2902292): Gtk-WARNING **: 18:14:27.053: cannot open display:

zy10110, 2025-05-21:
I've got the same problem  when I was Running Isaac-lab on Remote Server with RT X 4090 GPU [[omni.appwindow.plugin] Failed to acquire Windowing interface]

rolandying, 2025-05-26:
> Thank you for posting this. Have you checked this [tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/00_sim/launch_app.html) which refers to how to [set up livestream properly](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/manual_livestream_clients.html#isaac-sim-short-webrtc-streaming-client). Let us know if you need further help.

Hi @RandomOakForest 
I have the same problem with 2 3090 GPU when I running the example with livestream mode
`./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Ant-v0 --headless --livestream 2`
However, I can run the Isaac Sim 4.5.0 with livestream mode and the WebRTC works perfectly:
`./runheadless.sh -v`
This really bothering me, do u have any idea about the solutions?

stt-121106, 2025-06-09:
I encountered the same problem . When the visualization is not enabled, everything works fine ,but when it is turned on, the system crashes. Have u solved this question? If there is a way, please reply to me. This is really important for me

kellyguo11, 2025-06-09:
For the original question on the 5090, you are hitting a cuda version issue as Blackwell GPUs require cuda 12.8. You can update torch to 2.7+cu128 to resolve the cuda error. For the 4090 issues, the warnings shared shouldn't result in crashes. are you seeing any additional errors in the logs? also on multi-GPU machines, could you try forcing a single GPU with CUDA_VISIBLE_DEVICES?