isaac-sim/IsaacLab #2485: [Bug Report] Bug title

81578823, 2025-05-13:
The training shut down for no reasons with 50 series card  [bug].

### Describe the bug

The training shuts down for unknown reasons. Please help me. 
### Steps to reproduce
```bash
python scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-B-v0 --headless
```
and 
```bash
python scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --headless
```

### System Info

- Isaac Sim Version: 4.5.0
- OS: Ubuntu 20.04
- GPU: RTX5080
- CUDA:12.8
- GPU Driver: 570.144
- torch: 2.7.0

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

81578823, 2025-05-13:
[image]

81578823, 2025-05-13:
Sometimes, it just gets stuck and does not move on

[image]

RandomOakForest, 2025-05-14:
Thank you for posting this, the team will review it.

baijubaiju233, 2025-07-10:
Can you use the 50-series graphics cards for training? But when you open Isaac Sim 4.5 with a 50-series card, doesn't it blur objects? I think these two aren't compatible. I assumed that 50-series cards could only be used with Isaac Sim 5.0 paired with Isaac Lab 2.2. I’m really sorry for bringing up a topic unrelated to the issue, but I’m eager to know the answer.

AntoineRichard, 2026-08-27:
@81578823 huge thanks for circling back with the root cause — most people never report what finally fixed it.

For anyone landing here later: this was the host's `intel_pstate` CPU-frequency driver on an Intel Ultra 7-265K, not Isaac Lab. Booting with `intel_pstate=disable` (falling back to `acpi-cpufreq`) plus the `schedutil` governor stopped the crashes. Since there's nothing to change on our side we're closing this — but please reopen if you see the same symptom on current hardware and releases.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard