isaac-sim/IsaacLab #2490: [Bug Report] Bug title  When running isaac-sim.bat on Windows 11 to start isaac-simGUI, the command line reports an error: [Error] [omni.ext._impl.custom_importer] Failed to import python module omni.replicator.core. Error: No module named 'botocore'.

DarkKnightCyd, 2025-05-14:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

A clear and concise description of what the bug is.
I follow the [guide](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/install_workstation.html) to launch isaac-sim 4.5.0 through isaac-sim.selector.bat/isaac-sim.bat file but the cmd window reports several errors. All have one message in common:  `Error: No module named 'botocore'.`
### Steps to reproduce

Open cmd window at path "C:\Isaac", which is my extract root path  of  the isaacsim zip file. Running isaac-sim.selector.bat in cmd window. And the selector window shows up. Click start and infos emerging in an new cmd window while several errors shows up.

**error mesage 1:**
```text
[14,342ms] [Error] [omni.ext._impl.custom_importer] Failed to import python module omni.replicator.core. Error: No module named 'botocore'. Traceback:
Traceback (most recent call last):
  File "c:\issac/kit/kernel/py\omni\ext\_impl\custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "c:\issac\kit\python\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "c:/issac/extscache/omni.replicator.core-1.11.35+106.5.0.wx64.r.cp310/omni/replicator/core/__init__.py", line 3, in <module>
    from .scripts import (
  File "c:/issac/extscache/omni.replicator.core-1.11.35+106.5.0.wx64.r.cp310/omni/replicator/core/scripts/__init__.py", line 3, in <module>
    from . import (
  File "c:/issac/extscache/omni.replicator.core-1.11.35+106.5.0.wx64.r.cp310/omni/replicator/core/scripts/backends/__init__.py", line 21, in <module>
    from .s3 import S3Backend
  File "c:/issac/extscache/omni.replicator.core-1.11.35+106.5.0.wx64.r.cp310/omni/replicator/core/scripts/backends/s3.py", line 22, in <module>
    from botocore.exceptions import ClientError, NoCredentialsError
ModuleNotFoundError: No module named 'botocore'

 [14,342ms] [Error] [carb.scripting-python.plugin] Exception: Extension python module: 'omni.replicator.core' in 'c:\issac\extscache\omni.replicator.core-1.11.35+106.5.0.wx64.r.cp310' failed to load.

At:
  c:\issac/kit/kernel/py\omni\ext\_impl\_internal.py(213): startup
  c:\issac/kit/kernel/py\omni\ext\_impl\_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>

 [14,342ms] [Error] [omni.ext.plugin] [ext: omni.replicator.core-1.11.35] Failed to startup python extension.
```

**error mesage 2:**
```text
[14,517ms] [Error] [omni.ext._impl.custom_importer] Failed to import python module isaacsim.core.nodes. Error: No module named 'botocore'. Traceback:
Traceback (most recent call last):
  File "c:\issac/kit/kernel/py\omni\ext\_impl\custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "c:\issac\kit\python\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "c:/issac/exts/isaacsim.core.nodes/isaacsim/core/nodes/__init__.py", line 9, in <module>
    from .impl import BaseResetNode, BaseWriterNode, Extension, WriterRequest
  File "c:/issac/exts/isaacsim.core.nodes/isaacsim/core/nodes/impl/__init__.py", line 15, in <module>
    from .base_writer_node import BaseWriterNode, WriterRequest
  File "c:/issac/exts/isaacsim.core.nodes/isaacsim/core/nodes/impl/base_writer_node.py", line 14, in <module>
    import omni.replicator.core as rep
  File "c:/issac/extscache/omni.replicator.core-1.11.35+106.5.0.wx64.r.cp310/omni/replicator/core/__init__.py", line 3, in <module>
    from .scripts import (
  File "c:/issac/extscache/omni.replicator.core-1.11.35+106.5.0.wx64.r.cp310/omni/replicator/core/scripts/__init__.py", line 3, in <module>
    from . import (
  File "c:/issac/extscache/omni.replicator.core-1.11.35+106.5.0.wx64.r.cp310/omni/replicator/core/scripts/backends/__init__.py", line 21, in <module>
    from .s3 import S3Backend
  File "c:/issac/extscache/omni.replicator.core-1.11.35+106.5.0.wx64.r.cp310/omni/replicator/core/scripts/backends/s3.py", line 22, in <module>
    from botocore.exceptions import ClientError, NoCredentialsError
ModuleNotFoundError: No module named 'botocore'
 [Error] [carb.scripting-python.plugin] Exception: Extension python module: 'isaacsim.core.nodes' in 'c:\issac\exts\isaacsim.core.nodes' failed to load.

At:
  c:\issac/kit/kernel/py\omni\ext\_impl\_internal.py(213): startup
  c:\issac/kit/kernel/py\omni\ext\_impl\_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
 [Error] [omni.ext.plugin] [ext: isaacsim.core.nodes-2.1.4] Failed to startup python extension.
```
**error mesage 3:**
```text
 [15,272ms] [Error] [omni.ext._impl.custom_importer] Failed to import python module isaacsim.examples.interactive.tests. Error: No module named 'isaacsim.examples.interactive'. Traceback:
Traceback (most recent call last):
  File "c:\issac/kit/kernel/py\omni\ext\_impl\custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "c:\issac\kit\python\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'isaacsim.examples.interactive'

[Error] [carb.scripting-python.plugin] Exception: Extension python module: 'isaacsim.examples.interactive.tests' in 'c:\issac\exts\isaacsim.examples.interactive' failed to load.

At:
  c:\issac/kit/kernel/py\omni\ext\_impl\_internal.py(213): startup
  c:\issac/kit/kernel/py\omni\ext\_impl\_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
 [15,272ms] [Error] [omni.ext.plugin] [ext: isaacsim.examples.interactive-4.4.8] Failed to startup python extension.
```
...There's several messages I do not list out but has similar error 
```text
Error: No module named 'botocore'.
````
### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: Windows 11
- GPU: RTX 4070Ti Super
- CUDA:  12.6
- GPU Driver: 561.03 

### Additional context

Add any other context about the problem here.

### Checklist

- [ x] I have checked that there is no similar issue in the repo (**required**)
- [ x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ x] Criteria 1
- [ x] Criteria 2

RandomOakForest, 2025-05-19:
Thank you for posting this. Please follow the [installation steps here](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/index.html). You are likely hitting an issue with the setup of your environment.

DarkKnightCyd, 2025-05-20:
Thanks for your reply! I re-installed the software and the problem did not appear.

closed 2025-05-20 by DarkKnightCyd