isaac-sim/IsaacLab #2506: [Bug Report]  Isaac-sim GUI get stuck when launching from python venv with promt "isaacsim"

DarkKnightCyd, 2025-05-16:
### Describe the bug

The Isaac-sim GUI is stuck while loading assets. Specifically, I activate a python venv and run command "isaacsim" according to the [guide](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html), the cmd window shows no error message and the GUI shows up. But when I click to switch to the "isaac sim asset" tab, the GUI get stuck and the bottom right processing bar shows the asset is loaded from omniverse url. But I have confirmed that asset is downloaded to localhost and relative path have been modified. Besides, if  I run "Isaac-sim.bat" to launch the software the problem will not happen.

### Steps to reproduce

Open a cmd window and activate python venv. Run command isaacsim. The GUI shows up. Click "Isaac Sim Assets [Beta]" to switch to the tab. Then the bottom right part of GUI shows a process bar, but is stuck at 33%.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.5.0
- OS: Windows11
- GPU: [e.g. RTX 2060 Super]
- CUDA: 12.6
- GPU Driver:

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

Bardreamaster, 2025-05-19:
@DarkKnightCyd  It might be network error. Try access the url mentioned in your error message directly in your browser. If you can't download it, it means some network error make you can not access these assets. 

If it is network issue:
1. If you are in some kind of intranet, contact your network manager to add these address into white list.
2. If you are in China, try to change VPN or any other way to access amazonaws network source

DarkKnightCyd, 2025-05-19:
> [@DarkKnightCyd](https://github.com/DarkKnightCyd) It might be network error. Try to access the url mentioned in your error message directly in your browser. If you can't download it, it means some network error make you can not access these assets.
> 
> If it is network issue:
> 
> 1. If you are in some kind of intranet, contact your network manager to add these address into white list.
> 2. If you are in China, try to change VPN or any other way to access amazonaws network source

Thanks for your reply! I've downloaded assets to my local disk to avoid loading from network. This works well when I launch the app from "isaac-sim.bat" file. But why it cannot work with "isaacsim" command in python venv? Is there a way to make it loading resources from local? I've tried things like change asset loading path or cmd line args but they wont help.

RandomOakForest, 2025-06-11:
Thank you for following up. You may have to do a clean reinstall. I will close this issue for now. Please let us know if the issue persists. Thanks.

closed 2025-06-11 by RandomOakForest