isaac-sim/IsaacLab #2508: [Question] SDF mesh collider for mesh usd

Zyl-000, 2025-05-16:
### Question

Hi,
 I am working on a contact-rich object insertion task and I find out the comvex decomposition of the object is far to rough for simulation. I noticed there is a sdf mesh collider, and I want to know how to create a usd with mesh collider from mesh file.

RandomOakForest, 2025-05-20:
Thanks for posting this. This can be done using Omni Physics colliders, see [this doc for reference and examples](https://docs.omniverse.nvidia.com/kit/docs/omni_physics/latest/dev_guide/rigid_bodies_articulations/collision.html#createsdfcollider).

Zyl-000, 2025-05-20:
I noticed in https://docs.omniverse.nvidia.com/extensions/latest/ext_physics/rigid-bodies.html the Rigid-Body Collision Mesh Approximation are "convex hull",  "convex decomposition", "bounding sphere", "bounding box" and "sdf". But in mesh converter cfg of IsaacLab, except the first four type, there is another approximation called "meshSimplification", is this sdf?