isaac-sim/IsaacLab #2510: Imitation learning Environment generation

iamanshulchauhan1, 2025-05-16:
Hello,

I'm currently working on an imitation learning project and trying to create a custom task. However, it seems like the framework currently only supports task generation via the RL task constructor.

Is support for imitation learning task creation planned in the near future? I'm attempting to build a custom task using the existing example tasks as a reference, but the current structure is quite complex. A more streamlined or modular approach similar to what's available for RL tasks would be extremely helpful for imitation learning as well.

Thanks in advance

RandomOakForest, 2025-05-20:
Thanks for posting this. That is likely, as we are continuously working to add features to Isaac Lab and streamlining learning processes. I'll tag this as an enhancement for the team to discuss in the near future.

AntoineRichard, 2026-08-27:
Thanks for raising this @iamanshulchauhan1 — authoring a custom imitation-learning task shouldn't feel like a research project.

Isaac Lab does ship a full IL stack: the `isaaclab_mimic` package plus `scripts/imitation_learning/` (Mimic data generation, robomimic training, locomanipulation SDG). For your actual ask, the docs walk through it step by step in [Creating Your Own Isaac Lab Mimic Compatible Environments](https://isaac-sim.github.io/IsaacLab/release/3.0.0/source/overview/imitation-learning/teleop_imitation.html#creating-your-own-isaac-lab-mimic-compatible-environments), including the `ManagerBasedRLMimicEnv` methods to implement and the `SubTaskConfig` fields. If a specific step there is still unclear, open a focused issue and we'll dig in.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard