isaac-sim/IsaacLab #2513: [Bug Report] Error when running RSL RL Framework

LuckyCZD, 2025-05-17:
When running any training scripts on the RSL RL Framework, such as the one from the installation guide
`python scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Ant-v0 --headless`
it errors 
```
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.classic.ant.ant_env_cfg:AntEnvCfg
[INFO]: Parsing configuration from: isaaclab_tasks.manager_based.classic.ant.agents.rsl_rl_ppo_cfg:AntPPORunnerCfg
[INFO] Logging experiment in directory: D:\IsaacLab\logs\rsl_rl\ant
Exact experiment name requested from command line: 2025-05-17_11-09-37
Setting seed: 42
[INFO]: Base environment:
        Environment device    : cuda:0
        Environment seed      : 42
        Physics step-size     : 0.008333333333333333
        Rendering step-size   : 0.016666666666666666
        Environment step-size : 0.016666666666666666
2025-05-17 03:12:15 [164,505ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of C:\g\122538378\USD\pxr\usd\usd\stage.cpp -- In </World/ground/terrain>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/Environments/Grid/default_environment.usd@ for reference introduced by @anon:000001F98D148FE0:World0.usd@</World/ground/terrain>. (recomposing stage on stage @anon:000001F98D148FE0:World0.usd@ <000001F98D6EA550>)

[INFO]: Time taken for scene creation : 158.007192 seconds
Error executing job with overrides: []
Traceback (most recent call last):
  File "d:/isaaclab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "D:\IsaacLab\scripts\reinforcement_learning\rsl_rl\train.py", line 141, in main
    env = gym.make(args_cli.task, cfg=env_cfg, render_mode="rgb_array" if args_cli.video else None)
  File "C:\Users\User\.conda\envs\isaac\lib\site-packages\gymnasium\envs\registration.py", line 742, in make
    env = env_creator(**env_spec_kwargs)
  File "d:/isaaclab/source/isaaclab/isaaclab/envs/manager_based_rl_env.py", line 79, in __init__
    super().__init__(cfg=cfg)
  File "d:/isaaclab/source/isaaclab/isaaclab/envs/manager_based_env.py", line 130, in __init__
    self.scene = InteractiveScene(self.cfg.scene)
  File "d:/isaaclab/source/isaaclab/isaaclab/scene/interactive_scene.py", line 151, in __init__
    self._add_entities_from_cfg()
  File "d:/isaaclab/source/isaaclab/isaaclab/scene/interactive_scene.py", line 648, in _add_entities_from_cfg
    self._terrain = asset_cfg.class_type(asset_cfg)
  File "d:/isaaclab/source/isaaclab/isaaclab/terrains/terrain_importer.py", line 103, in __init__
    self.import_ground_plane("terrain")
  File "d:/isaaclab/source/isaaclab/isaaclab/terrains/terrain_importer.py", line 217, in import_ground_plane
    ground_plane_cfg.func(prim_path, ground_plane_cfg)
  File "d:/isaaclab/source/isaaclab/isaaclab/sim/spawners/from_files/from_files.py", line 145, in spawn_ground_plane
    collision_prim_path = prim_utils.get_prim_path(
  File "c:/users/user/.conda/envs/isaac/lib/site-packages/isaacsim/exts/isaacsim.core.utils/isaacsim/core/utils/prims.py", line 611, in get_prim_path
    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
2025-05-17 03:12:16 [164,774ms] [Warning] [omni.fabric.plugin] gFabricState->gUsdStageToSimStageWithHistoryMap had 1 outstanding SimStageWithHistory(s) at shutdown
2025-05-17 03:12:16 [164,864ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```
### Steps to reproduce

Following the pip installation on`https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html`

test`isaaclab.bat -p scripts\tutorials\00_sim\create_empty.py`, seems everything right, but when test `python scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Ant-v0 --headless`, it errors.
Also when test`isaaclab.bat -p scripts\demos\hands.py`, it errors

```
[13.484s] Simulation App Startup Complete
[13.545s] [ext: omni.physx.fabric-106.5.7] startup
2025-05-17 03:20:29 [127,207ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of C:\g\122538378\USD\pxr\usd\usd\stage.cpp -- In </World/defaultGroundPlane>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/Environments/Grid/default_environment.usd@ for reference introduced by @anon:0000022C853910F0:World0.usd@</World/defaultGroundPlane>. (recomposing stage on stage @anon:0000022C853910F0:World0.usd@ <0000022C81EE4430>)

Traceback (most recent call last):
  File "D:\IsaacLab\scripts\demos\hands.py", line 165, in <module>
    main()
  File "D:\IsaacLab\scripts\demos\hands.py", line 153, in main
    scene_entities, scene_origins = design_scene()
  File "D:\IsaacLab\scripts\demos\hands.py", line 69, in design_scene
    cfg.func("/World/defaultGroundPlane", cfg)
  File "d:/isaaclab/source/isaaclab/isaaclab/sim/spawners/from_files/from_files.py", line 145, in spawn_ground_plane
    collision_prim_path = prim_utils.get_prim_path(
  File "c:/users/user/.conda/envs/isaac/lib/site-packages/isaacsim/exts/isaacsim.core.utils/isaacsim/core/utils/prims.py", line 611, in get_prim_path
    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'
```

### System Info

- Commit: 21f7136
- Isaac Sim Version: 4.5
- OS: windows 10
- GPU: RTX4080SUPER
- CUDA: 11.7
- GPU Driver: 572.60

### Additional context

I also test on
- Commit: 21f7136
- Isaac Sim Version: 4.5
- OS: Ubuntu20.04
- GPU: RTX4080SUPER
- CUDA: 11.7
- GPU Driver: 550
Everything is right.  Is this version compatible with Windows?

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

zz7379, 2025-05-20:
same problem

hojae-io, 2025-05-21:
Same problem.
It seems this happens spontaneously. Sometime I can run, sometime it has this error (with exactly same code)

I guess the problem is not on the local code but on remote server? Like this part:

`Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.5/Isaac/Environments/Grid/default_environment.usd@ for reference introduced by @anon:000001F98D148FE0:World0.usd@</World/ground/terrain>. (recomposing stage on stage @anon:000001F98D148FE0:World0.usd@ <000001F98D6EA550>)`

closed 2025-06-11 by RandomOakForest