isaac-sim/IsaacLab #2523: [Bug Report] Bug title

calmdw, 2025-05-19:
Hi, I notice in the isaaclab_asset fold, there are configuration files for the robot, I noticed in the franka, UR10 and other robot arms are different. How these damping, stiffness parameters decided? Whats the support material?

Problem 1 about the stiffness and damping parameters. In the franka.py file the stiffness and damping are 80, 4 for arm joints, in universal_robots.py file the stiffness and damping are 800, 40 for arm joints. Theoretically, the franka should move faster with more vibration as smaller damping and stiffness, however, when I run the [Isaac-Reach-Franka-v0](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/reach/config/franka/joint_pos_env_cfg.py) and [Isaac-Reach-UR10-v0](https://github.com/isaac-sim/IsaacLab/blob/main/source/isaaclab_tasks/isaaclab_tasks/manager_based/manipulation/reach/config/ur_10/joint_pos_env_cfg.py). I notice the UR10 robot moves more faster with vibration.

Problem 2 about the effort_limit and velocity_limit, it seems they does not work, still in the franka-reach environment, the franka robot's movement looks same when I set the velocity_limit to be 0.2 and 2.

### System Info

Describe the characteristic of your environment:

- Commit: 
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `isaacsim 4.5.0'
- OS: [Ubuntu 22.04]
- GPU: [RTX 4090]
- CUDA: [12.2]
- GPU Driver: [535.183.01]

### Additional context

Add any other context about the problem here.

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] provide one reason
- [ ] one solution to solve the problem as the robot move too fast in the simulation, like punching into the target which is really dangerous in real experiment.

RandomOakForest, 2025-05-19:
Thank you for posting this. Could you please add a title and the system info to your issue? Thanks.

calmdw, 2025-05-19:
> Thank you for posting this. Could you please add a title and the system info to your issue? Thanks.

Sorry, I just notice it and add the name on it.

RandomOakForest, 2025-05-21:
These parameters are empirically set. You may find useful the [Gain Tuner Extension](https://docs.isaacsim.omniverse.nvidia.com/4.5.0/robot_setup/ext_isaacsim_robot_setup_gain_tuner.html) in the process of selecting your parameters. Would it be possible for you to use this tool to review your Problems 1 and 2? I will move this into a discussion for the team to follow up. Thank you.