isaac-sim/IsaacLab #2524: [Question] How do I safely remove prims in an event?

TCA166, 2025-05-19:
I have a `ManagerBasedRLEnvCfg` with an `EventCfg` that specifies a single event that ideally should reset the whole environment. In the function wrapped by the `EventTermCfg` I call `isaacsim.core.utils.stage.clear_stage`, followed by some calls that populate the scene anew. When that event fires, the following warning is shown immediately after the scene is cleared:
```
[Warning] [omni.physx.tensors.plugin] prim ... was deleted while being used by a shape in a tensor view class. The physics.tensors simulationView was invalidated.
```
followed by:
```
[Error] [omni.physx.tensors.plugin] Simulation view object is invalidated and cannot be used again to call setDofActuationForces
```
and this exception: 
```
Traceback (most recent call last):
  File "/projects/neuroforgelab/dev/nfl-lab-bridge/manager.py", line 212, in <module>
    run_manger(manager)
  File "/projects/neuroforgelab/dev/nfl-lab-bridge/manager.py", line 41, in run_manger
    scene.reset()
  File "/home/s481817/.conda/envs/env_isaaclab/lib/python3.10/site-packages/isaaclab/source/isaaclab/isaaclab/envs/manager_based_env.py", line 290, in reset
    self.scene.write_data_to_sim()
  File "/home/s481817/.conda/envs/env_isaaclab/lib/python3.10/site-packages/isaaclab/source/isaaclab/isaaclab/scene/interactive_scene.py", line 488, in write_data_to_sim
    articulation.write_data_to_sim()
  File "/home/s481817/.conda/envs/env_isaaclab/lib/python3.10/site-packages/isaaclab/source/isaaclab/isaaclab/assets/articulation/articulation.py", line 195, in write_data_to_sim
    self.root_physx_view.set_dof_actuation_forces(self._joint_effort_target_sim, self._ALL_INDICES)
  File "/home/s481817/.conda/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/extsPhysics/omni.physics.tensors/omni/physics/tensors/impl/api.py", line 1528, in set_dof_actuation_forces
    raise Exception("Failed to set DOF actuation forces in backend")
Exception: Failed to set DOF actuation forces in backend
```
From what I have managed to read (mainly [here](https://forums.developer.nvidia.com/t/delete-prim-from-simulation-and-keep-the-simulation-running/328770)), Isaac Sim won't like prims being deleted while the simulation is running. But how can I stop the simulation within an event? Is there an alternative method I'm missing? I would be most grateful for any help, I've been stuck on this for the better part of last two days.

Also I'm very sorry if this isn't the right place for this question.

RandomOakForest, 2025-05-20:
Thank you for posting this. The recommended approach is to pause simulation before deletion. That is, stop the simulation context using `world.stop()`, perform prim removal, then restart with `world.start()`.

TCA166, 2025-05-20:
Thank you for your response! If you would be so kind, as to point towards which class implements the `start` and `stop` methods and where can i obtain an instance of the `world` variable

RandomOakForest, 2025-06-11:
Try `from isaacsim.core.api.world import World`.

closed 2025-06-11 by RandomOakForest

h4ri98, 2025-06-16:
Can you please provide an example of how to use the world instance. I am trying to modify the scale of the prims on startup and reset, but creating a World instance from the env.sim context object and using stop and start does not seem to work. I am not getting any errors, but the training does not start
`def reset_environment(env: ManagerBasedRLEnv, env_ids: torch.Tensor) -> None:
   
    if(env.sim.is_playing()):
        world = World(env.sim)
        world.stop()
   *** Some processing to modify scale of certain prims in the scene ***
    world.start()
    `