isaac-sim/IsaacLab #2535: Error when try to run differential IK on UR10 robot

zakydzul, 2025-05-21:
**System : Ubuntu 22.04**

**Cuda version: 12.9**

**Isaac Sim: 4.5.0**

**Isaac Lab: Latest**

When I run the diff IK script using franka, it works like intended.
.`/isaaclab.sh -p scripts/tutorials/05_controllers/run_diff_ik.py --robot franka_panda --num_envs 128`
But when I use ur10, I got an error, possibly because the scene config is not correct.
.`/isaaclab.sh -p scripts/tutorials/05_controllers/run_diff_ik.py --robot ur10 --num_envs 128`

> Traceback (most recent call last):
>   File "/home/zaky/IsaacLab/scripts/tutorials/05_controllers/run_diff_ik.py", line 210, in <module>
>     main()
>   File "/home/zaky/IsaacLab/scripts/tutorials/05_controllers/run_diff_ik.py", line 199, in main
>     scene = InteractiveScene(scene_cfg)
>   File "/home/zaky/IsaacLab/source/isaaclab/isaaclab/scene/interactive_scene.py", line 151, in __init__
>     self._add_entities_from_cfg()
>   File "/home/zaky/IsaacLab/source/isaaclab/isaaclab/scene/interactive_scene.py", line 681, in _add_entities_from_cfg
>     asset_cfg.spawn.func(
>   File "/home/zaky/IsaacLab/source/isaaclab/isaaclab/sim/spawners/from_files/from_files.py", line 145, in spawn_ground_plane
>     collision_prim_path = prim_utils.get_prim_path(
>   File "/home/zaky/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/exts/isaacsim.core.utils/isaacsim/core/utils/prims.py", line 611, in get_prim_path
>     return prim.GetPath()
> AttributeError: 'NoneType' object has no attribute 'GetPath'

RandomOakForest, 2025-05-21:
Thank you for posting this. It seems the prim is missing. Where you able to install Isaac Lab and verify this installation?

zakydzul, 2025-05-22:
Yes, I was able to install the Isaac Lab and verify it's installed correctly, hence why I can run the diff IK with Franka robot.

> Thank you for posting this. It seems the prim is missing. Where you able to install Isaac Lab and verify this installation?

RandomOakForest, 2025-06-11:
Thank you for following up. The team will review this.

AntoineRichard, 2026-08-27:
Thanks for the traceback @zakydzul — a crash on a tutorial script is exactly the kind of thing we want to hear about.

We can't reproduce it on the current release. The failure was inside `spawn_ground_plane`, not the `--robot ur10` path (ground-plane spawning is identical for both robots), and it looks like a transient asset fetch — the default ground-plane USD resolves fine today. That code no longer calls `isaacsim.core.utils.prims.get_prim_path`; `source/isaaclab/isaaclab/sim/spawners/from_files/from_files.py` now raises an explicit `ValueError` if the collision prim is missing, so you'd get a clear message rather than an `AttributeError`. Please reopen with a log if it recurs.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard