isaac-sim/IsaacLab #2537: [Bug Report] Bug title

Antonello-Scaldaferri-adi, 2025-05-21:
On Windows 11, when running:

.\isaaclab.bat -p .\scripts\reinforcement_learning\rl_games\train.py --task Isaac-Factory-PegInsert-Direct-v0 --headless env.episode_length_s=5.0

I get:

Error parsing override 'env.episode_length_s'
missing EQUAL at '<EOF>'
See https://hydra.cc/docs/1.2/advanced/override_grammar/basic for details

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

How to solve this?

Thanks.

RandomOakForest, 2025-05-21:
Thank you for posting this. Could you share a more complete error log? It may be a previous error what causes this. Thanks.

kellyguo11, 2025-05-30:
could you try adding quotes around the additional argument? `"env.episode_length_s=5.0"`