isaac-sim/IsaacLab #2539: Error executing job with overrides: []

explore-code-world7, 2025-05-21:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

When I was running isaaclab with rsl_rl framework, got an error as below:

```bash
type = <class 'torch.Tensor'>, std = tensor([[1.0077, 1.0101, 1.0171,  ..., 0.9938, 0.9983, 0.9945],
        [1.0077, 1.0101, 1.0171,  ..., 0.9938, 0.9983, 0.9945],
        [1.0077, 1.0101, 1.0171,  ..., 0.9938, 0.9983, 0.9945],
        ...,
        [1.0077, 1.0101, 1.0171,  ..., 0.9938, 0.9983, 0.9945],
        [1.0077, 1.0101, 1.0171,  ..., 0.9938, 0.9983, 0.9945],
        [1.0077, 1.0101, 1.0171,  ..., 0.9938, 0.9983, 0.9945]],
       device='cuda:0', grad_fn=<ExpandBackward0>)
type = <class 'torch.Tensor'>, std = tensor([[nan, nan, nan,  ..., nan, nan, nan],
        [nan, nan, nan,  ..., nan, nan, nan],
        [nan, nan, nan,  ..., nan, nan, nan],
        ...,
        [nan, nan, nan,  ..., nan, nan, nan],
        [nan, nan, nan,  ..., nan, nan, nan],
        [nan, nan, nan,  ..., nan, nan, nan]], device='cuda:0',
       grad_fn=<ExpandBackward0>)
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/sky/Project/Robot/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 102, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/sky/Project/Robot/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 151, in main
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "/home/sky/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/rsl_rl/runners/amp_on_policy_runner.py", line 507, in learn
    ) = self.alg.update()
  File "/home/sky/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/rsl_rl/algorithms/amp_ppo.py", line 559, in update
    self.policy.act(
  File "/home/sky/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/rsl_rl/modules/actor_critic.py", line 134, in act
  File "/home/sky/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/torch/distributions/normal.py", line 73, in sample
    return torch.normal(self.loc.expand(shape), self.scale.expand(shape))
RuntimeError: normal expects all elements of std >= 0.0

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```
It can be seen that the last std became nan, but I don't know why now.

### Steps to reproduce
* None, since it's a code built myself, it's tough to upload the entire file box.
### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.5
- OS:  Ubuntu 22.04]
- GPU:  RTX 4060
- CUDA: 12.4
- GPU Driver: 550.144.03

### Additional context
None

### Checklist
- [ x] I have checked that there is no similar issue in the repo (**required**)
- [ x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

explore-code-world7, 2025-05-21:
Maybe it's caused by graphic memory excessive consuming?

RandomOakForest, 2025-05-22:
Thank you for posting this. The `nan`'s tensor points to some physics instability. We have a [doc for troubleshooting this](https://isaac-sim.github.io/IsaacLab/main/source/refs/troubleshooting.html). It may also have to do with your `_reset_idx` function, as in #1625 (see comments).

explore-code-world7, 2025-05-22:
> Thank you for posting this. The `nan`'s tensor points to some physics instability. We have a [doc for troubleshooting [...]

Yes, it seems it's caused by [simulation instability](https://docs.omniverse.nvidia.com/kit/docs/omni_physics/latest/dev_guide/guides/articulation_stability_guide.html).
Thank you a lot!

RandomOakForest, 2025-05-22:
Following up, in this case, it is an out-of-memory error, unfortunately.

explore-code-world7, 2025-05-25:
Another time, I got the problem again.
```bash
################################################################################
                     Learning iteration 3097/10000                      

                       Computation: 330526 steps/s (collection: 0.734s, learning 0.852s)
               Value function loss: 22878095257.6000
                    Surrogate loss: -0.0030
                           Entropy: 2.1839
                          Amp loss: 0.3976
                 Grad penalty loss: 0.1082
             Mean policy Pred loss: 0.3270
             Mean expert Pred loss: 0.6733
              Mean policy Accuracy: 0.9982
              Mean expert Accuracy: 1.0000
             Mean action noise std: 0.26
                       Mean reward: 24775.79
               Mean episode length: 896.92
                 Mean style reward: 8.0998
       Mean episode heading_reward: 29.9393
            Mean episode up_reward: 598.7917
         Mean episode alive_reward: 119.7583
      Mean episode progress_reward: 24.3280
         Mean episode actions_cost: 55.2876
          Mean episode energy_cost: 31.0732
    Mean episode dof_at_limit_cost: 1.8843
           Mean episode death_cost: 0.0003
        Mean episode hand_sup_cost: 0.0000
--------------------------------------------------------------------------------
                   Total timesteps: 1623719936
                    Iteration time: 1.59s
                      Time elapsed: 01:23:06
                               ETA: 03:05:11

amp_sample_num = 2000000
Error executing job with overrides: []
Error executing job with overrides: []
Error executing job with overrides: []
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/chenlei/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 102, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/chenlei/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 191, in main
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "/home/chenlei/IsaacLab/third_package/rsl_rl/rsl_rl/runners/amp_on_policy_runner.py", line 508, in learn
    ) = self.alg.update()
  File "/home/chenlei/IsaacLab/third_package/rsl_rl/rsl_rl/algorithms/amp_ppo.py", line 559, in update
    self.policy.act(
  File "/home/chenlei/IsaacLab/third_package/rsl_rl/rsl_rl/modules/actor_critic.py", line 122, in act
    return self.distribution.sample()
  File "/home/chenlei/isaacsim/exts/omni.isaac.ml_archive/pip_prebundle/torch/distributions/normal.py", line 73, in sample
    return torch.normal(self.loc.expand(shape), self.scale.expand(shape))
RuntimeError: normal expects all elements of std >= 0.0
```
* this is caused by excessive value function loss.
* I intend to reduce sim.dt to stablize training

closed 2025-05-25 by explore-code-world7