isaac-sim/IsaacLab #2542: [Question] Interactive simulation for Anymal velocity command

AvisP, 2025-05-22:
### Question

I am trying to creative an intteractive demo similar to the humanoid robot as shown in https://github.com/isaac-sim/IsaacLab/blob/main/scripts/demos/h1_locomotion.py but for Anymal on Rough terrain. I modified and have the following script. Although it is working properly, selecting a robot and pressing the the UP button it is proceeding forward. But when  and trying to go left or right by pressing LEFT and RIGHT buttons it is falling down.

I am guessing it has something to do with how the direction commands are specified here https://github.com/isaac-sim/IsaacLab/blob/6183e153f09f8740ad911bccb93ad4f65f48dbfe/scripts/demos/h1_locomotion.py#L134. I am not able to understand why for the H1 it was set like this and what it should be for the quadruped robot Anymal. Any tips would be appreciated.

```
import argparse
import os
import sys

sys.path.append(os.path.join(os.path.dirname(os.path.abspath(__file__)), "../.."))
import scripts.reinforcement_learning.rsl_rl.cli_args as cli_args  # isort: skip

from isaaclab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(
    description="This script demonstrates an interactive demo with the H1 rough terrain environment."
)
# append RSL-RL cli arguments
cli_args.add_rsl_rl_args(parser)
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""
import torch

import carb
import omni
from omni.kit.viewport.utility import get_viewport_from_window_name
from omni.kit.viewport.utility.camera_state import ViewportCameraState
from pxr import Gf, Sdf
from rsl_rl.runners import OnPolicyRunner

from isaaclab.envs import ManagerBasedRLEnv
from isaaclab.utils.math import quat_apply
from isaaclab.utils.pretrained_checkpoint import get_published_pretrained_checkpoint

from isaaclab_rl.rsl_rl import RslRlOnPolicyRunnerCfg, RslRlVecEnvWrapper

from isaaclab_tasks.manager_based.locomotion.velocity.config.anymal_c.rough_env_cfg import AnymalCRoughEnvCfg_PLAY
TASK = "Isaac-Velocity-Rough-Anymal-C-v0"
RL_LIBRARY = "rsl_rl"

class AnymalRoughDemo:
    """This class provides an interactive demo for the H1 rough terrain environment.
    It loads a pre-trained checkpoint for the Isaac-Velocity-Rough-H1-v0 task, trained with RSL RL
    and defines a set of keyboard commands for directing motion of selected robots.

    A robot can be selected from the scene through a mouse click. Once selected, the following
    keyboard controls can be used to control the robot:

    * UP: go forward
    * LEFT: turn left
    * RIGHT: turn right
    * DOWN: stop
    * C: switch between third-person and perspective views
    * ESC: exit current third-person view"""

    def __init__(self):
        """Initializes environment config designed for the interactive model and sets up the environment,
        loads pre-trained checkpoints, and registers keyboard events."""
        agent_cfg: RslRlOnPolicyRunnerCfg = cli_args.parse_rsl_rl_cfg(TASK, args_cli)
[... 119 lines cut ...]
        camera_state = ViewportCameraState(self.camera_path, self.viewport)
        eye = Gf.Vec3d(camera_pos[0].item(), camera_pos[1].item(), camera_pos[2].item())
        target = Gf.Vec3d(base_pos[0].item(), base_pos[1].item(), base_pos[2].item() + 0.6)
        camera_state.set_position_world(eye, True)
        camera_state.set_target_world(target, True)

def main():
    """Main function."""
    demo_anymal = AnymalRoughDemo()
    obs, _ = demo_anymal.env.reset()
    while simulation_app.is_running():
        # check for selected robots
        demo_anymal.update_selected_object()
        with torch.inference_mode():
            action = demo_anymal.policy(obs)
            obs, _, _, _ = demo_anymal.env.step(action)
            print("Action ", action)
            print("Obserrvation ", obs)
            # overwrite command based on keyboard input
            obs[:, 9:13] = demo_anymal.commands

if __name__ == "__main__":
    main()
    simulation_app.close()
```

RandomOakForest, 2025-05-23:
Thanks for posting this. Could you verify that the input of the command manager is correct, verify the mapping for the keys is ok?

AvisP, 2025-05-26:
@RandomOakForest Thanks for your suggestion. By `command_manager` are you referring to these line?
```
self.env.unwrapped.command_manager.reset([self._previous_selected_id])
self.commands[:, 0:3] = self.env.unwrapped.command_manager.get_command("base_velocity")
```
The values get set in `self.commands` when I am selecting a bot and pressing buttons they get set on this line to predefined fixed values

https://github.com/isaac-sim/IsaacLab/blob/6183e153f09f8740ad911bccb93ad4f65f48dbfe/scripts/demos/h1_locomotion.py#L148                  

I managed to stabilize the movement a bit by changing these lines. I don't fully understand what values to put where and they take values from 1 to -1. Is it possible to pass a vector of floating points instead of binding the buttons to a fixed vector? Kindly advise, I am new to this

```
T = 1.0
R = 0.5
F = 0.8
self._key_to_control = {
    "UP": torch.tensor([T, 0.0, 0.0, 0.0], device=self.device),
    "DOWN": torch.tensor([-T, 0.0, 0.0, -R], device=self.device),
    "LEFT": torch.tensor([T, 0.6, -0.6, 0], device=self.device),
    "RIGHT": torch.tensor([T, -0.6, 0.6, 0], device=self.device),
    "ZEROS": torch.tensor([0.0, 0.0, 0.0, 0.0], device=self.device),
}
```