isaac-sim/IsaacLab #2543: [Proposal] Please add support the root link alias

celestialdr4g0n, 2025-05-22:
### Proposal

Add support for configurable root link aliasing in simulation environments.
Enable users to specify a custom root link alias to "override" the default root_link_pose selection, particularly in scenarios involving mobile manipulators where the desired root link may differ from what the simulator chooses.

### Motivation

In current implementations (e.g., Isaac Sim), the simulator automatically selects a root link—often not aligned with the user’s expectations or robot design intent. For mobile manipulators, it’s common to treat the chassis link as the root, but Isaac Sim may default to a manipulator joint, which cause trouble for resetting states of the environments

### Alternatives

My work around this issue. But I think an alias method will make it cleaner.

```
root_state = self.rb.data.default_root_state.clone()[env_ids]
wrong_base_base_inv_pos, wrong_base_base_inv_quat = subtract_frame_transforms(
    self.rb.data.body_pos_w[env_ids, self.env.base_link_idx, :3],
    self.rb.data.body_quat_w[env_ids, self.env.base_link_idx, :4],
    self.rb.data.root_pos_w[env_ids, :3],
    self.rb.data.root_quat_w[env_ids, :4]
)
root_state_correct_pos_w, root_state_correct_quat_w = combine_frame_transforms(
    root_state[:, :3],
    root_state[:, 3:7],
    wrong_base_base_inv_pos,
    wrong_base_base_inv_quat
)
root_state_correct = root_state.clone()
root_state_correct[:, :3] = root_state_correct_pos_w + self.scene.env_origins[env_ids]
root_state_correct[:, 3:7] = root_state_correct_quat_w
root_vel_zeros = torch.zeros_like(root_state[:, 7:])
self.rb.write_root_pose_to_sim(root_state_correct[:, 0:7], env_ids=env_ids)
self.rb.write_root_velocity_to_sim(root_vel_zeros, env_ids=env_ids)
```
### Additional context

"Left blank"

### Checklist

- [x] I have checked that there is no similar issue in the repo.

### Acceptance Criteria

- Users can specify a root_link_alias that "overrides" the default root link.
- The simulator correctly maps state reads/writes through the alias.
- Backward compatibility is preserved (no alias → current behavior).
- Documentation added explaining the usage and constraints of this feature.

RandomOakForest, 2025-05-22:
Thank you for submitting this proposal. The team will review it.

StafaH, 2026-09-27:
Thanks @celestialdr4g0n! For a floating-base mobile manipulator, move `UsdPhysics.ArticulationRootAPI` from its current prim onto the chassis rigid-body prim before simulation initialization. This explicitly selects the chassis as the actual root, avoiding the need for an alias. See the [NVIDIA documentation](https://docs.omniverse.nvidia.com/kit/docs/omni_physics/110.0/dev_guide/rigid_bodies_articulations/articulations.html#usd-hierarchy-articulation-root).

Closing with this asset-authoring solution. Please reopen this issue if it does not work for your robot.

closed 2026-09-27 by StafaH