isaac-sim/IsaacLab #2546: [Question] Accessing agent config environment initialization

ozhanozen, 2025-05-22:
### Question

I would like to access the agent config from the environment config for manager-based environments. More specifically, I intend to adapt the Curriculum terms based on the `num_steps_per_env` of the rsl_rl agent config. However, as the environment is decoupled from the agent, I didn't see an easy way to do this.

Do you have any ideas regarding how to do that?

Thanks in advance.

RandomOakForest, 2025-05-22:
Thank you for posting this. Maybe an environment-agent interface would work?
```
class CustomCurriculumManager(CurriculumManager):
    def set_agent_config(self, agent_cfg: object):
        self._agent_cfg = agent_cfg

    def update(self):
        # Access num_steps_per_env from agent config
        steps = self._agent_cfg.runner.num_steps_per_env
        # Update curriculum parameters
        ...
```

ozhanozen, 2025-05-23:
> Thank you for posting this. Maybe an environment-agent interface would work?
> 
> ```
> class CustomCurriculumManager(CurriculumManager):
>     def set_agent_config(self, agent_cfg: object):
>         self._agent_cfg = agent_cfg
> 
>     def update(self):
>         # Access num_steps_per_env from agent config
>         steps = self._agent_cfg.runner.num_steps_per_env
>         # Update curriculum parameters
>         ...
> ```
> 
> You may have to also modify `train.py` to have access to the agent config.

Thanks for the reply @RandomOakForest!

I agree, those changes seem necessary. However, `CustomCurriculumManager` (or `CurriculumManager`) only has access to what `ManagerBasedRLEnv` has, which currently does not include `agent_cfg`. So, in addition to what you suggest, I think we would also need to modify `ManagerBasedRLEnv` to accept `agent_cfg` as an argument first. Would you agree?

These are fairly fundamental design changes, but they could be valuable as a general feature for others a well. Do you think it would make sense to open a PR for this?

RandomOakForest, 2025-06-11:
Thank you, that seems to be correct. A PR would be welcomed. Thanks!

StafaH, 2026-09-20:
Thanks @ozhanozen for raising this. We recommend keeping `ManagerBasedRLEnv` independent of the agent configuration: different RL libraries represent rollout settings differently, and play or standalone environment use has no agent config at all.

For a curriculum expressed in training iterations, set the derived environment-step threshold after resolving both configs and before creating the environment:

```python
env_cfg.curriculum.my_term.params["num_steps"] = (
    curriculum_iteration * agent_cfg.num_steps_per_env
)
env = gym.make(task_name, cfg=env_cfg)
```

Alternatively, define the rollout length once as a shared constant used by both configs. Existing curricula can then continue using `env.common_step_counter` without adding an environment-agent interface. We are closing this as not planned because the use case can be handled at config-composition time without coupling the environment to a particular runner.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-09-20 by StafaH