isaac-sim/IsaacLab #2547: [Question] Is it possible to assign compliant contact parameters to instanced envs?

likesweetie, 2025-05-22:
### Question

Hello.

I'm currently trying to train a quadruped robot with highly deformable foot pads. Which should be considered during its training senarios. So I decide to give compliant contact parameters to foot material for simulating such compliant behavior. However, It seems that assigning physics materials to instanced prim is not possible on current Isaac Lab framework...

What I need to do is randomize wide range of material stiffness and damping for robot foot and make robot to experience many types of compliant contact on training. So just giving single material properties to terrain is insufficient.

Any solution or suggestion for this?

[image]

RandomOakForest, 2025-05-23:
Thank you for posting this. You are correct, currently there is a limitation when using instanceable assets as you mention, but the team is already working on this for a future release. A workaround would to use the UI and use the checkbox for enabling/disabling instanceable on the prims.

likesweetie, 2025-05-23:
> Thank you for posting this. You are correct, currently there is a limitation when using instanceable assets as you mention, but the team is already working on this for a future release. A workaround would to use the UI and use the checkbox for enabling/disabling instanceable on the prims.

Thank you for the confirmation. I going to try that option.

closed 2025-05-23 by likesweetie