isaac-sim/IsaacLab #2555: [Question] Loading Custom 6-DOF Robot USD in Isaac Lab via ArticulationCfg - Hangs at sim.reset()

tyc333, 2025-05-23:
### Question

I'm encountering a bug while trying to load a custom 6-DOF robot into Isaac Lab.

My setup involves a 6-axis robotic arm, similar in structure to a UR5, with its mesh defined by STL files and its kinematics by a URDF file.

Here's a summary of what works and where I'm facing a problem:

What Works:

- I can successfully import the URDF file into Isaac Sim using the Import Wizard.
- The imported robot functions correctly within Isaac Sim:
- I can control it using Lula inverse kinematics.
- I can also achieve the same control using the Isaac Sim Python API.
- I can export a USD file of this robot from Isaac Sim, and this USD file appears to be valid and loads correctly back into Isaac Sim.

The Problem:
I am trying to integrate this robot into an Isaac Lab environment using the python api ArticulationCfg. To do this, I modified one of the Isaac Lab demo files (specifically, a file similar to arms.py in the demos) by replacing the USD path of an existing robot with the path to my custom robot's USD file (the one exported from Isaac Sim).

When I run this modified Isaac Lab script:

- The Isaac Sim UI launches successfully, and I see the crosshairs indicating the interface has loaded.
- However, neither my robot nor the simulation environment appears.
- The simulation seems to hang at the sim.reset() call.
- Crucially, there are no error messages in the console, which makes debugging very difficult.

Troubleshooting Steps & Observations:

- If I use the exact same Isaac Lab code but replace my custom robot's USD path with a path to a downloaded UR5 USD file, the simulation loads and runs correctly (the UR5 robot and environment appear).
- This leads me to suspect there might be an issue with my robot's URDF or STL files, or perhaps how the USD is generated from them, that specifically affects Isaac Lab's loading mechanism via ArticulationCfg, even though Isaac Sim's standard import and simulation handle it without issues.

I am quite puzzled by this discrepancy, as the asset works in Isaac Sim but not in Isaac Lab under these specific conditions. The lack of error messages has made it challenging to pinpoint the cause, and I've been struggling with this for several days due to limited documentation on such specific issues.

Could you provide any guidance or suggestions on what might be causing this behavior or what I should look for in my URDF/STL/USD files to ensure compatibility with Isaac Lab's ArticulationCfg? Are there any specific USD properties or structures that Isaac Lab expects which might be missing or incorrectly configured for my custom asset?
I have provided my urdf and stl files, and I'm using isaacsim4.5 and isaaclab2.0.2.

Thanks!

[robo_test.zip](https://github.com/user-attachments/files/20408019/robo_test.zip)

RandomOakForest, 2025-05-29:
Thank you for posting this. Indeed, without an error message it would be really hard to spot what's going on. Have you checked you're not running out of memory? or maybe it was a one-time issue that may resolve after relaunching the apps?

tyc333, 2025-05-29:
> Thank you for posting this. Indeed, without an error message it would be really hard to spot what's going on. Have you checked you're not running out of memory? or maybe it was a one-time issue that may resolve after relaunching the apps?

Yes and No. I doubt it's a memory problem since I have 32GB of RAM and the arms.py demo runs fine. I've also restarted both the PC and the app multiple times with no change. My URDF file is attached above. Could you please try importing it on your end with the Isaac Lab to see if it succeeds?

RandomOakForest, 2025-06-11:
Following up, there seems it is the URDF that is at fault. I'll move this to our _Discussions_ for follow up.