isaac-sim/IsaacLab #2558: [Question] Documentation on creating yaml files for learning configurations.

toenails6, 2025-05-23:
### Question

From what I can gather, Isaac Lab configures learning parameters based on the corresponding RL libraries. 
But upon looking at Isaac-Velocity-Rough-Unitree-Go1-v0, I noticed that the given SKRL yaml configurations don't quite fit the official SKRL documentations. 

For one, the 'separate' parameter is set to false, but there are separate definitions for policy and value networks in the yaml. 
I also could not find any documentation on the 'separate' parameter in the official SKRL docs. 

SKRL official docs also specify layers and activations in multiple lines, which I can understand how the given yaml maps to that, but it would be better to have some form of explanation and/or docs on what the given yaml means. 

Do we have any other guides to creating learning configurations?

RandomOakForest, 2025-05-30:
Thanks for posting this. @Toni-SM for vis.

Toni-SM, 2025-06-02:
Hi @toenails6 

> For one, the 'separate' parameter is set to false, but there are separate definitions for policy and value networks in the yaml

PPO agent/algorithm uses 1 stochastic (`policy`) and 1 deterministic (`value`) function approximator (see [skrl's PPO Models](https://skrl.readthedocs.io/en/latest/api/agents/ppo.html#models) 
In the Isaac Lab `.yaml` the `policy` and `value` class definitions are defined using such classes: `GaussianMixin` (stochastic and continuos, for the `policy`) and `DeterministicMixin` (deterministic, for the `value`).
Now, the `separate` config refers how the `network` definition will be handled between the 2 models. If `False` (shared model), the policy `network` definition will be used for the `value` too (no matter what definition it has for this).

https://github.com/isaac-sim/IsaacLab/blob/dca126ac9d280e44adf7e9e5bdf7adcc6a3ed765/source/isaaclab_tasks/isaaclab_tasks/manager_based/locomotion/velocity/config/go1/agents/skrl_rough_ppo_cfg.yaml#L6-L29

> it would be better to have some form of explanation and/or docs on what the given yaml means.

Yup, while you can find in skrl's docs documentation about defining the `network` configuration for each model (see [skrl's Model instantiators](https://skrl.readthedocs.io/en/latest/api/utils/model_instantiators.html)), there is not already documentation about full `.yaml` configuration/structure apart for the existing files. I am planning a [skrl's Runner](https://skrl.readthedocs.io/en/latest/api/utils/runner.html) docs update to clarify it.

toenails6, 2025-09-30:
For anyone that might read this thread, checkout the [SKRL Runner documentation](https://skrl.readthedocs.io/en/latest/api/utils/runner.html) for the answer to this question:
https://skrl.readthedocs.io/en/latest/api/utils/runner.html