isaac-sim/IsaacLab #2562: The policy.pt file cannot be deployed in IsaacSim 4.5

TomJimmy-33, 2025-05-24:
The policy. pt obtained from reinforcement learning using IsaacLab 2.1.0 does not contain the comstants.pkl , resulting in an error when deploying on IsaacSim 4.5.0. How can I solve this problem？ I tried to use **rsl_rl** and **skrl**， results are the same. I also checked the versions of PyTorch in my IsaacSim and IsaacLab environments, and they are consistent.

Error: PytorchStreamReader failed locationg file constants.pkl: file not found.

RandomOakForest, 2025-05-29:
Thank you for posting this. Could you share the error log you get? We would need to see the earlier error messages as well. Thanks.

TomJimmy-33, 2025-05-30:
Error log:
2025-05-30 03:12:59 [9,736ms] [Error] [omni.kit.app._impl] [py stderr]: Traceback (most recent call last):
2025-05-30 03:12:59 [9,736ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/username/.local/share/ov/pkg/isaac-sim-4.5.0/standalone_examples/api/isaacsim.robot.policy.examples/sz_standalone.py", line 72, in <module>
2025-05-30 03:12:59 [9,736ms] [Error] [omni.kit.app._impl] [py stderr]:     sz = szFlatTerrainPolicy(
2025-05-30 03:12:59 [9,736ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/username/anaconda3/envs/env_isaacsim/lib/python3.10/site-packages/isaacsim/exts/isaacsim.robot.policy.examples/isaacsim/robot/policy/examples/robots/sz.py", line 48, in __init__
2025-05-30 03:12:59 [9,736ms] [Error] [omni.kit.app._impl] [py stderr]:     self.load_policy(
2025-05-30 03:12:59 [9,736ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/username/anaconda3/envs/env_isaacsim/lib/python3.10/site-packages/isaacsim/exts/isaacsim.robot.policy.examples/isaacsim/robot/policy/examples/controllers/policy_controller.py", line 73, in load_policy
2025-05-30 03:12:59 [9,736ms] [Error] [omni.kit.app._impl] [py stderr]:     self.policy = torch.jit.load(file)
2025-05-30 03:12:59 [9,736ms] [Error] [omni.kit.app._impl] [py stderr]:   File "/home/username/anaconda3/envs/env_isaacsim/lib/python3.10/site-packages/torch/jit/_serialization.py", line 165, in load
2025-05-30 03:12:59 [9,736ms] [Error] [omni.kit.app._impl] [py stderr]:     cpp_module = torch._C.import_ir_module_from_buffer(
2025-05-30 03:12:59 [9,737ms] [Error] [omni.kit.app._impl] [py stderr]: RuntimeError: istream reader failed: getting the current position.

Recently, I found that the policy. pt file obtained using rsl_rl contains the constants. pkl file, while the best agent. pt file obtained using skrl does not contain the constants. pkl file. The former can be deployed in IsaacSim, while the latter cannot be deployed

RandomOakForest, 2025-06-11:
Following up, could you post steps to reproduce this? Thanks