isaac-sim/IsaacLab #2566: Dismatch between joint_pos with joint_pos_limits

explore-code-world7, 2025-05-25:
### Question

Basic questions, related to robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

RandomOakForest, 2025-05-29:
Thank you for posting this. The values output are rather large. Could you verify that at the start from loading the robot, you are also seeing these kind of values? It would be helpful to know how many steps after launching the simulation you are getting this.

explore-code-world7, 2025-06-02:
* Further, I add debugging info in `source/isaaclab_tasks/isaaclab_tasks/direct/locomotion/locomotion_env.py`

```python
    def _apply_action(self):
        print("before")
        print((self.robot.data.joint_pos[0] <= self.robot.data.joint_pos_limits[0,:,0]).sum(),\
              (self.robot.data.joint_pos[0] >= self.robot.data.joint_pos_limits[0,:,1]).sum())
        print(self.robot.data.joint_pos[0])
        forces = self.action_scale * self.joint_gears * self.actions
        self.robot.set_joint_effort_target(forces, joint_ids=self._joint_dof_idx)
```
and `source/isaaclab/isaaclab/envs/direct_rl_env.py`
```python
    def step(self, action: torch.Tensor) -> VecEnvStepReturn:
        for _ in range(self.cfg.decimation):
            self._sim_step_counter += 1
            # set actions into buffers
            self._apply_action()
            # set actions into simulator
            self.scene.write_data_to_sim()
            # simulate
            self.sim.step(render=False)
            # render between steps only if the GUI or an RTX sensor needs it
            # note: we assume the render interval to be the shortest accepted rendering interval.
            #    If a camera needs rendering at a faster frequency, this will lead to unexpected behavior.
            if self._sim_step_counter % self.cfg.sim.render_interval == 0 and is_rendering:
                self.sim.render()
            # update buffers at sim dt
            self.scene.update(dt=self.physics_dt)
            print("after")
            print((self.robot.data.joint_pos[0] <= self.robot.data.joint_pos_limits[0, :, 0]).sum(), \
                  (self.robot.data.joint_pos[0] >= self.robot.data.joint_pos_limits[0, :, 1]).sum())
            print(self.robot.data.joint_pos[0])
```

* the result shows that apply of first action and sim.step produces joint_poses exceeding joint_limits

[image]

RandomOakForest, 2025-06-12:
Following up, it seems you may be hitting an issue introduced earlier. Could you upgrade to the current v2.1.0 of Isaac Lab if you haven't? [See these notes for reference](https://isaac-sim.github.io/IsaacLab/main/source/refs/release_notes.html#migration-guide).

AntoineRichard, 2026-08-27:
Thanks for the detailed numbers @explore-code-world7, and sorry for the delay. We weren't able to get to the bottom of this while it was open, and the report now predates several releases, so we're closing it out.

For what it's worth, joint positions around 9335 rad against limits of `[0, 1.7]` most likely mean the solve diverged rather than a limits-reporting bug — `joint_pos_limits` reports the configured drive limits, and positions aren't hard-clamped once a solve blows up, so the converted USD's masses/inertias and your actuator gains are the first things we'd check. Happy to reopen with a repro on a current release.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard