isaac-sim/IsaacLab #2567: Can't instantiate abstract class RslRlVecEnvWrapper with abstract method get_privileged_observations

Ericshuai1, 2025-05-25:
### Question

Basic questions, related to robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).
When I run the command
./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --headless
report errors
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/eric/IsaacLab/source/isaaclab_tasks/isaaclab_tasks/utils/hydra.py", line 101, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/eric/IsaacLab/scripts/reinforcement_learning/rsl_rl/train.py", line 164, in main
    env = RslRlVecEnvWrapper(env, clip_actions=agent_cfg.clip_actions)
TypeError: Can't instantiate abstract class RslRlVecEnvWrapper with abstract method get_privileged_observations

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace
Thank you very much for your answer

RandomOakForest, 2025-05-29:
Thanks for posting this. Could you try updating the RSL RL library to the latest version? Thanks.

vineetpasumarti, 2025-06-06:
Were you able to resolve this by updating the RSL RL library? I'm having an identical issue.

closed 2025-06-12 by RandomOakForest