isaac-sim/IsaacLab #2581: [Proposal] Wrappers for importing multiple urdf

kbkartik, 2025-05-27:
### Proposal

The goal here is to define wrappers for importing multiple URDFs similar to how we already have [MultiUsdFileCfg](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sim.spawners.html#isaaclab.sim.spawners.wrappers.MultiUsdFileCfg)

### Motivation

Popular object datasets such as [dexenv](https://huggingface.co/datasets/taochenshh/dexenv) often are list of URDFs files. Hence, wrappers to import multiple URDFs into IsaacLab will be useful for future users.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

- [] Import a bunch of URDFs from [dexenv](https://huggingface.co/datasets/taochenshh/dexenv) dataset and check if random objects' are spawning in parallel environments.

RandomOakForest, 2025-05-29:
Thanks for posting this proposal. The team will review it.

AntoineRichard, 2026-08-27:
Thanks for the proposal, @kbkartik — bulk-importing a URDF dataset is a real workflow.

There's no `MultiUrdfFileCfg` and we're not planning one, because the supported path already covers it in two steps: convert your URDFs once with `UrdfConverter` (or `scripts/tools/convert_urdf.py`), then spawn the generated USDs with `MultiUsdFileCfg` / `MultiAssetSpawnerCfg` for per-environment variety. Keeping conversion as an explicit offline step also means the converted assets are inspectable and reusable across runs.

Closing on that basis — please reopen if the offline conversion step is impractical for your dataset and we'll reconsider.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard