isaac-sim/IsaacLab #2588: [Question] How to directly query the validity (if in collision) of a robot state without changing the robot to that state in the simulation progress?

ProNeverFake, 2025-05-28:
### Question

Does Isaac Sim/Lab provide APIs for checking if a robot state is valid (not in collision with other objects)?

Currently I'm trying to integrate motion planning algorithms into isaacsim (like OMPL). Since these algorithms are very much sample-based methods, a `isStateValid()` method is necessary for discarding all invalid samples, leaving only valid samples to connect them into a path. An intuitive solution would be letting the robot be at the sample state and do collision checks. This could possibly work. But it would be great if I can directly query the validity of a robot state without changing the state of my current simulation progress.

Is there a feature or method helpful for this problem? Is a direct query in this case possible? I noticed the curobo implementation but found out it is optimization-based so it seems to be not directly helpful for this case.

RandomOakForest, 2025-05-30:
Thank you for posting this. This type of validity check is implemented in Isaac Lab through [contact sensors](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sensors.html). For example:
```
# Enable contact sensors during robot spawning
robot_cfg = RigidObjectSpawnerCfg(
    activate_contact_sensors=True,
    # ... other configs
)

# Create sensor for robot links
contact_sensor = ContactSensor(
    prim_path="/World/envs/env_.*/Robot/.*",
    filter_prim_paths_expr=["/World/obstacles/*"],
    update_period=0.01
)

# During simulation loop
if contact_sensor.data.net_contact_forces.norm() > threshold:
    invalidate_robot_state()
```

ProNeverFake, 2025-05-30:
> Thank you for posting this. This type of validity check may be implemented in Isaac Lab through [contact sensors](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.sensors.html). For example:
> 
> ```
> # Enable contact sensors during robot spawning
> robot_cfg = RigidObjectSpawnerCfg(
>     activate_contact_sensors=True,
>     # ... other configs
> )
> 
> # Create sensor for robot links
> contact_sensor = ContactSensor(
>     prim_path="/World/envs/env_.*/Robot/.*",
>     filter_prim_paths_expr=["/World/obstacles/*"],
>     update_period=0.01
> )
> 
> # During simulation loop
> if contact_sensor.data.net_contact_forces.norm() > threshold:
>     invalidate_robot_state()
> ```

Thanks for sharing your insight! This can definitely help if we traverse all the sampled robot states, setting the robot states and checking possible collisions by reading the contact sensors on our robot bodies. 

I'll say this is currently the only implementation I can think of to apply packages like OMPL to IsaacSim/lab. Not efficient enough, but it can very possibly work. BTW, I also suggest a better tutorial for Curobo. Some code in the tutorial is out-of-date for version 4.5, and the tutorial is not detailed enough to spare developers' integration effort. :)

closed 2025-06-12 by RandomOakForest