isaac-sim/IsaacLab #2590: Dead code in CircularBuffer?

Nate711, 2025-05-28:
It looks like this line of code isn't doing anything: https://github.com/isaac-sim/IsaacLab/blob/988cadf207549d4da94672c9a34e1ea38eefc301/source/isaaclab/isaaclab/utils/buffers/circular_buffer.py#L132 

Additionally, that code could be simplified with something like:
```python
# Check for batches with zero pushes and initialize all values in batch to first append
zero_pushes_mask = self._num_pushes == 0
if torch.any(zero_pushes_mask):
    fill_ids = zero_pushes_mask.nonzero().squeeze(1)
    self._buffer[:, fill_ids, :] = data.to(self._device)[fill_ids].unsqueeze(0).expand(self.max_length, -1, -1)
```

RandomOakForest, 2025-05-30:
Thanks for posting this. It seems you are correct. The team will review this.

Mayankm96, 2025-06-04:
Thanks for pointing this out. Made an MR for it.

commit d63e58f9 (2025-06-06): Simplifies buffer validation check for CircularBuffer (#2617)

# Description

As pointed out in the reported issue, there seem to be some expensive
`tolist()` operations inside the circular buffer class, that aren't
necessary. This MR simplifies the code.

Fixes #2590

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./isaaclab.sh --format`
- [ ] I have made corresponding changes to the documentation
- [ ] My changes generate no new warnings
- [ ] I have added tests that prove my fix [... cut]
closed 2025-06-06 by kellyguo11