isaac-sim/IsaacLab #2601: Flow Simulation does not work in isaac sim when started the robot training

NishuT26, 2025-06-01:
I am trying to run an Omniverse Flow fire simulation within Isaac Sim. The fire effect works perfectly when I press the main “Play” button in the interactive Isaac Sim UI, but it remains completely static (it does not start or animate) when I start Isaac Sim with python script for robot training.

My assumption is that the Isaac Lab script takes manual control of the simulation timeline, and this manual timeline does not automatically include the Flow simulation. While observing the Flow Monitor (Window>Flow>Flow Monitor), when i start the Isaac sim normally and click on play, the values of Active Block Count was continuously being updated, but when for the simulation where Isaac lab was in effect, at that time the Active Block Count was 0 for long time, then it directly increased to 6 and then stopped updating.

NishuT26, 2025-06-01:
Starting isaac sim with XR rendering mode fixed the issue.

closed 2025-06-01 by NishuT26

george-nehma, 2025-11-20:
Hi @NishuT26,

I am trying to add this fire animation to my IsaacLab environment as well. Could you show how you added the fire to your simulation in the python script?