isaac-sim/IsaacLab #2605: [carb.scenerenderer-rtx.plugin] Failed to execute RenderGraph on device 0.

explore-code-world7, 2025-06-02:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

* When I running env with camera sensors with excessive num_envs, 
```bash
./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py  --task Isaac-H1-Walk-v0  --info  --headless  --enable_cameras  --num_envs 128
```
* a bug reports as follows:
```bash
2025-06-02 07:25:13 [33,654ms] [Error] [rtx.raytracing.plugin] Failed to allocate ParameterBlock resources
2025-06-02 07:25:13 [33,654ms] [Error] [gpu.foundation.plugin] Unable to allocate descriptor sets.
2025-06-02 07:25:13 [33,654ms] [Error] [rtx.raytracing.plugin] Failed to allocate ParameterBlock resources
2025-06-02 07:25:16 [36,921ms] [Error] [gpu.foundation.plugin] Failed to update params for RenderOp AOV Converter: extract disocclusion mask. Will not execute this or subsequent RenderGraph operations. Aborting RenderGraph execution
2025-06-02 07:25:16 [36,925ms] [Error] [carb.scenerenderer-rtx.plugin] Failed to execute RenderGraph on device 0. Error Code: 7
2025-06-02 07:25:16 [36,925ms] [Warning] [omni.hydra] Rendering failed.
2025-06-02 07:25:16 [36,926ms] [Error] [omni.usd] HydraEngine::render failed to end the compute graph: error code 6
```
* it may be caused by excessive gpu consuming, but the output of nvidia-smi shows gpu consuming is far less than gpu storage.

[image]

* besides, when decreasing num_envs to 64 
```bash
./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py  --task Isaac-H1-Walk-v0  --info  --headless  --enable_cameras  --num_envs 64
```
* the code run normally
[image]

* with gpu consuming less than 8G

[image]

* even double env nums, the gpu consuming should < 16G, and the former output also shows 8G which << single gpu storage=24G

* it's really confusing~

### Steps to reproduce

* left out

### System Info

- Isaac Sim Version: 4.5
- OS:  [Ubuntu 22.04]
- GPU:  RTX 4060
- CUDA: 12.4
- GPU Driver: 550.144.03

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

explore-code-world7, 2025-06-03:
> Thank you for posting this. Your log reports an earlier error on _GLInteropContext_ that may be triggering this. Are you running this locally or on the cloud? If the latter, be sure you have `export DISPLAY=:0`, which has resolved this issue in some cases. Also, please try using the [recommended driver](https://docs.isaacsim.omniverse.nvidia.com/latest/installation/requirements.html#driver-requirements).

* Thank your timely reaply.
* the former javascript runs well 

```python
./isaaclab.sh -p scripts/reinforcement_learning/rsl_rl/train.py  --task Isaac-H1-Walk-v0  --info  --headless  --enable_cameras  --num_envs 64
```
with `GLInteropContext` Error, so the problem shouldn't be it.
* I tried `export DISPLAY=:0`(before) , and problems remains the same.

RandomOakForest, 2025-06-12:
Following up, the driver you are using is still not the recommended one. You may also want to try [tiled rendering](https://isaac-sim.github.io/IsaacLab/main/source/overview/core-concepts/sensors/camera.html).