isaac-sim/IsaacLab #2619: [Bug Report] Non-Physical Explosive Motion When D6 Joint Reaches breakForce Threshold​​

Benoooochen, 2025-06-04:
### Describe the bug

When a D6 Joint reaches its configured breakForce threshold, the connected rigid bodies exhibit violent, non-physical motion (e.g., explosive separation, uncontrolled rotation, or interpenetration) instead of a stable separation. This violates energy conservation principles and compromises simulation fidelity

### Steps to reproduce

[image]

```

-->

### System Info

isaacsim 4.5   latest version of isaaclab

RandomOakForest, 2025-06-06:
Thank you for submitting this. Could you share code and USDs we could use to reproduce this observation? Can you also post your Isaac Lab version and platform specs? Thank you.

AntoineRichard, 2026-08-27:
@Frankkkkchen thanks for the write-up on the explosive motion when a D6 joint hits its `breakForce` threshold.

Break-force response is resolved inside the PhysX solver rather than in Isaac Lab code, so a minimal repro is what we need before we can route it properly. Without one we couldn't take it further at the time, and the report now predates several releases. Closing as stale — please reopen with a repro on a current release and we'll pick it back up.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard