isaac-sim/IsaacLab #2639: [Bug Report] Failed to modify a joint's initial position (ArticulationCfg.InitialStateCfg)

Gariscat, 2025-06-08:
Command:
``

I was trying to modify the initial configuration of the cabinet in "Isaac-Franka-Cabinet-Direct-v0". Modify the position of cabinet was successful. However, when I modified the initial state of the joint, it seems like nothing happened.

First case - modify the position of cabinet:
`init_state=ArticulationCfg.InitialStateCfg(
            ### pos=(0.0, 0, 0.4),
            pos=(0.0, 0, 10),   # mutable (checked)
            rot=(0.1, 0.0, 0.0, 0.0),
            joint_pos={
                "door_left_joint": 0.0,
                "door_right_joint": 0.0,
                "drawer_bottom_joint": 0.0,
                ### "drawer_bottom_joint": 0.4, # NOTHING HAPPENED???
                "drawer_top_joint": 0.0,
            },
        ),`
[image]

Second case - modify the joint state:
`init_state=ArticulationCfg.InitialStateCfg(
            pos=(0.0, 0, 0.4),
            ### pos=(0.0, 0, 10),   # mutable (checked)
            rot=(0.1, 0.0, 0.0, 0.0),
            joint_pos={
                "door_left_joint": 0.0,
                "door_right_joint": 0.0,
                ### "drawer_bottom_joint": 0.0,
                "drawer_bottom_joint": 0.4, # NOTHING HAPPENED???
                "drawer_top_joint": 0.0,
            },
        ),`

[image]

Is this a possible bug of how ArticulationCfg.InitialStateCfg interacts with Isaac Sim?

kellyguo11, 2025-06-09:
Could it be the task is still resetting the transforms to some default configuration? Are you looking at the UI before or after the first reset in the environment?

Gariscat, 2025-06-09:
> Could it be the task is still resetting the transforms to some default configuration? Are you looking at the UI before or after the first reset in the environment?

Fascinating point! It turns out the `_reset_idx` method changed everything back to zeros. After trying to modify the joint_pos in `_reset_idx`, it worked!

Thanks a lot!

closed 2025-06-09 by RandomOakForest