isaac-sim/IsaacLab #2644: [Question] How to install a nightly version of Isaac Lab through pip

MARMOTatZJU, 2025-06-09:
### Question

Hello Team Isaac Lab. I am working on a project whose dependency relies on a nightly version of IsaacLab. Given that installing through pip is more efficienty and more recommended, I would like to know if it is possible and how to install a nightly version of Isaac Lab through pip.

Thanks in advance!

RandomOakForest, 2025-06-09:
Thanks for posting this. This may not be possible at the moment. Does your project depends on Isaac Sim nightly builds as well?

MARMOTatZJU, 2025-06-10:
@RandomOakForest Also I would like to know which way to install is recommended by Team Isaac Lab. Is it the way via `./isaaclab.sh --install` or the way via `pip install isaaclab ...`?

closed 2025-06-29 by RandomOakForest