isaac-sim/IsaacLab #2678: [Bug Report] Apple vision pro cannot properly render Isaac Lab environment

liuj1, 2025-06-12:
### Describe the bug

I'm trying to teleoperate an Isaac Lab robot with apple vision pro with CloudXR runtime. While in AR mode, the teleoperation data can be properly received (Robotic arms and hands in Isaac Lab move according to tracking data published by apple vision pro), but the virtual graphical environment cannot be properly rendered on apple vision pro (only shows some noises).

### Steps to reproduce

I followed the official instruction from https://isaac-sim.github.io/IsaacLab/main/source/how-to/cloudxr_teleoperation.html#run-isaac-lab-with-the-cloudxr-runtime to launch Isaac Lab and CloudXR runtime docker on my host machine as well as building and running Isaac XR Teleop Sample Client on apple vision pro.

I also tried the alternative option provided by the official instruction which runs CloudXR runtime container with docker and Isaac Lab as a local process.

```
docker run -it --rm --name cloudxr-runtime \
    --user $(id -u):$(id -g) \
    --gpus=all \
    -e "ACCEPT_EULA=Y" \
    --mount type=bind,src=$(pwd)/openxr,dst=/openxr \
    -p 48010:48010 \
    -p 47998:47998/udp \
    -p 47999:47999/udp \
    -p 48000:48000/udp \
    -p 48005:48005/udp \
    -p 48008:48008/udp \
    -p 48012:48012/udp \
    nvcr.io/nvidia/cloudxr-runtime:0.1.0-isaac
```

```
./isaaclab.sh -p scripts/environments/teleoperation/teleop_se3_agent.py \
    --task Isaac-PickPlace-GR1T2-Abs-v0 \
    --teleop_device dualhandtracking_abs \
    --enable_pinocchio
```

On CloudXR runtime, I receive error messages:

During launching
```
[NvNetworkCommon] (E)[2025-06-12 04:08:23,579]=04:08:23=*00000*{7442471588647581478}<NvNetworkCommon>  Network Address To String: invalid parameters 0x7ffcae4c26c0, 0x7ffcae4c26e0, 48
```

During streaming
```
RROR [cxr_server_log] [RtspHandshake] (C)[2025-06-12 04:13:18,474]=04:13:18=*00000*{14926899404732402132}<RtspHandshake>    Out of buffer space when trying to write format type spacer.
ERROR [cxr_server_log] [RtspHandshake] (C)[2025-06-12 04:13:18,474]=04:13:18=*00000*{14926899404732402132}<RtspHandshake>    Out of buffer space when trying to write format type.
ERROR [cxr_server_log] [NvscStreamingSession] (E)[2025-06-12 04:13:18,850]=04:13:18=*00000*{5398884187204415560}<NvscStreamingSes> Failed converting transit selection values : Invalid input.
```

### System Info

- Isaac Sim Version: [4.5.0]
- OS: [Ubuntu 22.04]
- GPU: [4 x RTX 4090]
- CUDA: [12.8]
- GPU Driver: [570.133.20]

RandomOakForest, 2025-06-12:
Thank you for posting this. The error at launch may be ignored. The streaming errors point to a CloudXR setup issue. The errors during streaming are from when you used Option 1 or Option 2 in the [docs](https://isaac-sim.github.io/IsaacLab/main/source/how-to/cloudxr_teleoperation.html#run-isaac-lab-with-the-cloudxr-runtime)?

liuj1, 2025-06-13:
> Thank you for posting this. The error at launch may be ignored. The streaming errors point to a CloudXR setup issue. The errors during streaming are from when you used Option 1 or Option 2 in the [...]

Thank you for the reply. The streaming errors are reported using option 2. I also tried option 1, while errors are not reported in the terminal, the rendering on AVP is still unsuccessful.

AntoineRichard, 2026-08-27:
@liuj1 thanks for reporting the AVP rendering artifacts, and for the setup details — they helped match this up.

We're consolidating on #3674, which describes the same CloudXR symptom (tracking works, robot moves, headset shows mosaic/noise) and has more corroboration, including a Quest 3 report. Your report came first, so the context here isn't lost — please follow and add to #3674. Note the CloudXR teleop docs have been rewritten since (Isaac Teleop, uv, single-container Docker), so it's worth retrying against the current instructions first. Reopen here if your symptom turns out to be distinct.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard