isaac-sim/IsaacLab #2693: [Bug Report]Because the thumbnail couldn't be opened. isaac sim

li269150, 2025-06-13:
[image]
[image]
Every time I open Isaac Lab, it gets stuck for a long time due to the inability to load PNG images. The assets have been placed locally. By using ./isaacsim.sh to open Isaac Sim, it can be opened and there is also a thumbnail. However, using ./isaaclab.sh to open training or a blank Isaac Sim will get stuck in the Isaac Sim dialog box due to the thumbnail issue. Please help me solve this problem.

li269150, 2025-06-13:
[image]The window I opened with the file "./isaacsim.sh" can successfully open the local "assest" assets without any issues.

li269150, 2025-06-14:
Isaac sim 4.5 isaac lab 2.0, ubuntu 20 locally installed

---- Replied Message ----
| From | Pablo ***@***.***> |
| Date | 06/14/2025 21:29 |
| To | isaac-sim/IsaacLab ***@***.***> |
| Cc | li269150 ***@***.***>,
Author ***@***.***> |
| Subject | Re: [isaac-sim/IsaacLab] [Bug Report]Because the thumbnail couldn't be opened. isaac sim (Issue #2693) |
RandomOakForest left a comment (isaac-sim/IsaacLab#2693)

Thank you for posting this. Could you share the Isaac Lab and Isaac Sim versions you are working with, and what process did you follow to install them? Thanks.

RandomOakForest, 2025-08-01:
Following up, I'll close this for now, as you may have resolved it already. If you still see this in Isaac Sim 5.0, please open a new _bug report_. Follow #3021 to install the latest versions of the tools. Thanks.

closed 2025-08-01 by RandomOakForest