isaac-sim/IsaacLab #2705: [Bug Report] Unable to run tutorial scripts with xcb error

jeffsonyu, 2025-06-16:
Hi,
I am trying to run the scripts in tutorials as:
```python
python scripts/tutorials/00_sim/create_empty.py
```
Failed due to:
```shell
[INFO][AppLauncher]: Using device: cuda:0
[INFO][AppLauncher]: Loading experience file: /home/jeffsonyu/IsaacLab/apps/isaaclab.python.kit
[Info] [carb] Logging to file: /home/jeffsonyu/isaacsim/kit/logs/Kit/Isaac-Sim/4.5/kit_20250616_113227.log
2025-06-16 03:32:27 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-06-16 03:32:27 [1ms] [Warning] [omni.ext.plugin] [ext: rendering_modes] Extensions config 'extension.toml' doesn't exist '/home/jeffsonyu/IsaacLab/apps/rendering_modes' or '/home/jeffsonyu/IsaacLab/apps/rendering_modes/config'
[0.036s] [ext: omni.kit.async_engine-0.0.1] startup
[0.320s] [ext: omni.metrics.core-0.0.1] startup
[0.320s] [ext: omni.client.lib-1.0.0] startup
[0.341s] [ext: omni.blobkey-1.1.2] startup
[0.341s] [ext: omni.stats-1.0.1] startup
[0.342s] [ext: omni.datastore-0.0.0] startup
[0.346s] [ext: omni.client-1.2.2] startup
[0.350s] [ext: omni.ujitso.default-1.0.0] startup
[0.351s] [ext: omni.hsscclient-1.1.1] startup
2025-06-16 03:32:28 [340ms] [Warning] [omni.datastore] OmniHub is inaccessible
[0.354s] [ext: omni.gpu_foundation.shadercache.vulkan-1.0.0] startup
[0.356s] [ext: omni.assets.plugins-0.0.0] startup
[0.357s] [ext: omni.gpu_foundation-0.0.0] startup
[0.362s] [ext: carb.windowing.plugins-1.0.0] startup
[0.368s] [ext: omni.kit.renderer.init-0.0.0] startup
2025-06-16 03:32:28 [386ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 24 belongs to.
2025-06-16 03:32:28 [386ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 25 belongs to.
2025-06-16 03:32:28 [386ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 26 belongs to.
2025-06-16 03:32:28 [386ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 27 belongs to.
2025-06-16 03:32:28 [386ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 28 belongs to.
2025-06-16 03:32:28 [386ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 29 belongs to.
2025-06-16 03:32:28 [386ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 30 belongs to.
2025-06-16 03:32:28 [386ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 31 belongs to.

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.230.02    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4080 SUPER    | Yes: 0 |     | 16376   MB | 10de      | 0          |
|     |                                  |        |     |            | 2702      | 777ed0c3.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-139-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: 13th Gen Intel(R) Core(TM) i9-13900K
| Cores: 24 | Logical Cores: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 128431 | Free Memory: 121057
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
[0.861s] [ext: omni.kit.pipapi-0.0.0] startup
[0.862s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.862s] [ext: omni.pip.compute-1.4.5] startup
[0.863s] [ext: omni.pip.cloud-1.1.7] startup
[0.865s] [ext: omni.isaac.core_archive-2.3.3] startup
[0.865s] [ext: omni.materialx.libs-1.0.6] startup
[0.871s] [ext: omni.kit.loop-isaac-1.2.1] startup
[0.871s] [ext: omni.kit.test-1.1.2] startup
[0.905s] [ext: omni.isaac.ml_archive-2.1.2] startup
[0.906s] [ext: omni.kit.telemetry-0.5.1] startup
[xcb] Unknown sequence number while appending request
[xcb] You called XInitThreads, this is not your fault
[xcb] Aborting, sorry about that.
[xcb] Extra reply data still left in queue
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
Fatal Python error: Segmentation fault

Thread 0x00007fd72d463740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/core/__init__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/home/jeffsonyu/isaacsim/extscache/omni.kit.telemetry-0.5.1+d02c707b.lx64.r.cp310/omni/kit/telemetry/_itelemetry.py", line 6 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/home/jeffsonyu/isaacsim/extscache/omni.kit.telemetry-0.5.1+d02c707b.lx64.r.cp310/omni/kit/telemetry/__init__.py", line 8 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1050 in _gcd_import
  File "/home/jeffsonyu/miniconda3/envs/isaac/lib/python3.10/importlib/__init__.py", line 126 in import_module
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/ext/_impl/custom_importer.py", line 76 in import_module
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 94 in _import_public
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 187 in _custom_importer
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 211 in startup
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 328 in startup_extension
  File "PythonExtension.cpp::startup()", line 2 in <module>
  File "/home/jeffsonyu/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 427 in _start_app
  File "/home/jeffsonyu/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 212 in __init__
  File "/home/jeffsonyu/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 780 in _create_app
  File "/home/jeffsonyu/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 131 in __init__
  File "/home/jeffsonyu/IsaacLab/scripts/tutorials/00_sim/create_empty.py", line 29 in <module>

Extension modules: psutil._psutil_linux, psutil._psutil_posix (total: 2)
Segmentation fault (core dumped)
```

The installation of IsaacSim has been verified. Could you help me with this ? I really appreciate it.

### System Info

Describe the characteristic of your environment:

- Commit: 28ed0bb
- Isaac Sim Version: 4.5.0-rc.36
- OS: Ubuntu 20.04
- GPU: RTX 4080
- CUDA: 12.1
- GPU Driver: 535.230.02

jeffsonyu, 2025-06-16:
I try to use headless mode to run the scripts and it still occurs the same error.
```[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Using device: cuda:0
[INFO][AppLauncher]: Loading experience file: /home/jeffsonyu/IsaacLab/apps/isaaclab.python.headless.kit
[Info] [carb] Logging to file: /home/jeffsonyu/isaacsim/kit/logs/Kit/Isaac-Sim/4.5/kit_20250616_113753.log
2025-06-16 03:37:53 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-06-16 03:37:53 [1ms] [Warning] [omni.ext.plugin] [ext: rendering_modes] Extensions config 'extension.toml' doesn't exist '/home/jeffsonyu/IsaacLab/apps/rendering_modes' or '/home/jeffsonyu/IsaacLab/apps/rendering_modes/config'
2025-06-16 03:37:54 [301ms] [Warning] [omni.usd_config.extension] Enable omni.materialx.libs extension to use MaterialX
2025-06-16 03:37:54 [409ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 24 belongs to.
2025-06-16 03:37:54 [409ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 25 belongs to.
2025-06-16 03:37:54 [409ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 26 belongs to.
2025-06-16 03:37:54 [409ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 27 belongs to.
2025-06-16 03:37:54 [409ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 28 belongs to.
2025-06-16 03:37:54 [409ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 29 belongs to.
2025-06-16 03:37:54 [409ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 30 belongs to.
2025-06-16 03:37:54 [409ms] [Error] [omni.platforminfo.plugin] failed to find the package that core 31 belongs to.
[xcb] Unknown sequence number while appending request
[xcb] You called XInitThreads, this is not your fault
[xcb] Aborting, sorry about that.
[xcb] Extra reply data still left in queue
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
Fatal Python error: Segmentation fault

Thread 0x00007f57e3d95740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1050 in _gcd_import
  File "/home/jeffsonyu/miniconda3/envs/isaac/lib/python3.10/importlib/__init__.py", line 126 in import_module
  File "/home/jeffsonyu/isaacsim/extscache/omni.usd.libs-1.0.1+d02c707b.lx64.r.cp310/pxr/Tf/__init__.py", line 89 in PreparePythonModule
  File "/home/jeffsonyu/isaacsim/extscache/omni.usd.libs-1.0.1+d02c707b.lx64.r.cp310/pxr/Gf/__init__.py", line 30 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1050 in _gcd_import
  File "/home/jeffsonyu/miniconda3/envs/isaac/lib/python3.10/importlib/__init__.py", line 126 in import_module
  File "/home/jeffsonyu/isaacsim/extscache/omni.usd.libs-1.0.1+d02c707b.lx64.r.cp310/pxr/Tf/__init__.py", line 89 in PreparePythonModule
  File "/home/jeffsonyu/isaacsim/extscache/omni.usd.schema.audio-0.0.0+d02c707b.lx64.r.cp310/OmniAudioSchema/__init__.py", line 46 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1050 in _gcd_import
  File "/home/jeffsonyu/miniconda3/envs/isaac/lib/python3.10/importlib/__init__.py", line 126 in import_module
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/ext/_impl/custom_importer.py", line 76 in import_module
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 94 in _import_public
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 187 in _custom_importer
  File "/home/jeffsonyu/isaacsim/kit/kernel/py/omni/ext/_impl/_internal.py", line 211 in startup
[... cut ...]
  File "/home/jeffsonyu/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 427 in _start_app
  File "/home/jeffsonyu/isaacsim/exts/isaacsim.simulation_app/isaacsim/simulation_app/simulation_app.py", line 212 in __init__
  File "/home/jeffsonyu/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 780 in _create_app
  File "/home/jeffsonyu/IsaacLab/source/isaaclab/isaaclab/app/app_launcher.py", line 131 in __init__
  File "/home/jeffsonyu/IsaacLab/scripts/tutorials/00_sim/create_empty.py", line 29 in <module>
Segmentation fault (core dumped)
```

chizhang1626, 2025-06-16:
Hello, I encountered the same issue on Ubuntu 20.04（RTX 4080, Driver Version: 535.183.01 ）. May I ask if you eventually solved it?

fan-ziqi, 2025-06-17:
Same question

```bash
(lab2) ubuntu@ubuntu-4090:~$ isaacsim
Loading user config located at: '/home/ubuntu/.local/share/ov/data/Kit/Isaac-Sim Full/4.5/user.config.json'
[Info] [carb] Logging to file: /home/ubuntu/.nvidia-omniverse/logs/Kit/Isaac-Sim Full/4.5/kit_20250617_184754.log
2025-06-17 10:47:54 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.031s] [ext: omni.kit.async_engine-0.0.1] startup
[0.123s] [ext: omni.metrics.core-0.0.1] startup
[0.123s] [ext: omni.client.lib-1.0.0] startup
[0.139s] [ext: omni.blobkey-1.1.2] startup
[0.139s] [ext: omni.stats-1.0.1] startup
[0.139s] [ext: omni.datastore-0.0.0] startup
[0.142s] [ext: omni.client-1.2.2] startup
[0.196s] [ext: omni.ujitso.default-1.0.0] startup
[0.200s] [ext: omni.hsscclient-1.1.1] startup
[0.205s] [ext: omni.gpu_foundation.shadercache.vulkan-1.0.0] startup
[0.207s] [ext: omni.assets.plugins-0.0.0] startup
[0.207s] [ext: omni.gpu_foundation-0.0.0] startup
[0.211s] [ext: carb.windowing.plugins-1.0.0] startup
[0.215s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.100       | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090          | Yes: 0 |     | 24564   MB | 10de      | 0          |
|     |                                  |        |     |            | 2684      | 16760a40.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-131-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: Intel(R) Core(TM) i9-14900KF
| Cores: 24 | Logical Cores: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 64106 | Free Memory: 9093
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2045
|---------------------------------------------------------------------------------------------|
[0.576s] [ext: omni.kit.pipapi-0.0.0] startup
[0.577s] [ext: omni.materialx.libs-1.0.6] startup
[0.581s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.581s] [ext: omni.usd.config-1.0.5] startup
[0.583s] [ext: omni.gpucompute.plugins-0.0.0] startup
[0.583s] [ext: omni.usd.libs-1.0.1] startup
[0.643s] [ext: omni.mdl-55.0.1] startup
[0.654s] [ext: omni.iray.libs-0.0.0] startup
[0.656s] [ext: omni.mdl.neuraylib-0.2.10] startup
[0.657s] [ext: omni.kit.usd.mdl-1.0.7] startup
[0.713s] [ext: omni.kit.telemetry-0.5.1] startup
[xcb] Unknown sequence number while appending request
[xcb] You called XInitThreads, this is not your fault
[xcb] Aborting, sorry about that.
[xcb] Extra reply data still left in queue
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
段错误 (核心已转储)
```

ShineSea, 2025-11-14:
> Fixed in https://forums.developer.nvidia.com/t/xcb-error-occurs-when-launching-isaacsim4-5-in-ubuntu20-04/336491/3?u=fan-ziqi
> 
> ```
> conda config --add channels conda-forge
> conda config --set channel_priority strict
> ```

Well done! Thanks!