isaac-sim/IsaacLab #2712: [Question] Define observation and state space for asymmetric actor-critic learning

cowcnovo, 2025-06-16:
I have a Direct workflow based project setup where I am training a robot arm to pick up a cube. As observations, I am giving the robot joint positions and velocities and depth data from a camera. I want to give privileged information to the agent, specifically the position of the object. I read in the Isaac Lab docs, [state_space](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/isaaclab.envs.html#isaaclab.envs.DirectRLEnvCfg.state_space), that the state_space should be modified/is useful for asymmetric learning. My questions are:
1) Since I have an asymmetric approach where the value network gets different inputs than the policy network, how should I modify the space definitions or other configurations to ensure it is correctly defined?
2) I could not find examples in the Isaac Lab GitHub repos for asymmetric learning. If there are some, could you please point me to them?

Here are my observations method:

```
def _get_observations(self) -> dict:
    object_pos_w = self.object.data.root_pos_w[:, :3]  # Object position in world frame
    robot_pos_w = self.ur5e.data.root_state_w[:, :3]  # Robot base position in world frame
    robot_quat_w = self.ur5e.data.root_state_w[:, 3:7]  # Robot base orientation in world frame

    # Transform object position to robot's root frame
    object_pos_b, _ = subtract_frame_transforms(robot_pos_w, robot_quat_w, object_pos_w)

    # Concatenate robot state and object position for observations
    robot_state = torch.cat(
        [
            self.ur5e_joint_pos[:, self.arm_joints_ids],
            self.ur5e_joint_vel[:, self.arm_joints_ids],
            self.ur5e_joint_pos[:, self.gripper_joints_ids],
            self.ur5e_joint_vel[:, self.gripper_joints_ids],
        ],
        dim=-1,
    )

    camera_data = self.camera.data.output["depth"]

    observations = {
        "policy": {
            "robot_state": robot_state,
            "camera_data": camera_data,
            "object_position": object_pos_b
        }
    }

    return observations
```

Here is my current SKRL yaml:

```
models:
  separate: False
  policy:  # see gaussian_model parameters
    class: GaussianMixin
    clip_actions: False
    clip_log_std: True
    min_log_std: -20.0
    max_log_std: 2.0
    initial_log_std: 0.5
    network:
    - name: features_extractor
      input: permute(OBSERVATIONS["camera_data"], (0, 3, 1, 2)) # PyTorch NHWC -> NCHW. Warning: don't permute for JAX since it expects NHWC
      layers:
        - conv2d: {out_channels: 32, kernel_size: 8, stride: 4, padding: 0}
        - conv2d: {out_channels: 64, kernel_size: 4, stride: 2, padding: 0}
        - conv2d: {out_channels: 64, kernel_size: 3, stride: 1, padding: 0}
        - flatten
      activations: relu
    - name: net
      input: concatenate([features_extractor, OBSERVATIONS["robot_state"]])
      layers: [512, 256, 128]
      activations: elu
    output: ACTIONS
  value:  # see deterministic_model parameters
    class: DeterministicMixin
    clip_actions: False
    network:
      - name: net
        input: concatenate(OBSERVATIONS["object_position"], OBSERVATIONS["robot_state"])
        layers: [256, 128, 64]
        activations: elu
    output: ONE
```

And finally, here are my space definitions:

```
# - spaces definition
action_space = 7
observation_space = {
    "robot_state": 16,
    "camera_data": [camera_width, camera_height, 1],
    "object_position": 3,
}
state_space = 0
```

ozhanozen, 2025-06-16:
Hi @cowcnovo,

1. In your script, you have:
```
observations = {
        "policy": {
            "robot_state": robot_state,
            "camera_data": camera_data,
            "object_position": object_pos_b
        }
    }
```

by using "policy" as a key, you are setting these observations for your policy network, and if there is no other key, same for your critic network (symmetric). If you create another entry with the "critic" key, then you would make your actor/critic asymmetric, e.g.:
```
observations = {
        "policy": {
            "robot_state": robot_state,
            "camera_data": camera_data,
        },
        "critic": {
            "object_position": object_pos_b
        }
    }
```
Have a look at this for more details:
https://github.com/isaac-sim/IsaacLab/blob/28ed0bb6713081fece9945904b08742c54633c2c/source/isaaclab/isaaclab/envs/common.py#L84-L96

2. Actor-critic is the name of the algorithmic pattern. The actor uses a policy network, which takes "policy" observations and outputs either a probability distribution for actions or simply actions. The critic uses a value network, which takes "critic" observations as input and outputs “how good” that state (or state–action pair) is. About "policy-value", I am not sure, maybe a network that has a shared part, and has two heads, one giving actions/probabilities, the other giving state-value? (in comparison to using separate networks)

3. `FactoryEnv` should be using an asymmetric actor-critic.

cowcnovo, 2025-06-16:
Hi @ozhanozen,

Thank you for your message, that cleared a lot of doubts! As a follow-up, how would I then use the critic key in my SKRL yaml file to define the networks? From what I understand from reading the sources you provided, I could just use:

` OBSERVATIONS["object_position"]`

which will directly take it from the "critic" group of the observations?