isaac-sim/IsaacLab #2739: Question: Is There an Official or Recommended Way to Convert Isaac Lab HDF5 Data to LeRobot Schema?

devedgar, 2025-06-19:
Hi,
I'm using Isaac Lab to collect demonstration data in HDF5 format for Groot N1.5 training tasks.

Is there an official or recommended tool/workflow to convert Isaac Lab HDF5 data to the LeRobot-compatible schema for GR00T N1.5 training?

If any examples or documentation exist, especially for humanoid robot data, it would be greatly appreciated.
Thank you!

RandomOakForest, 2025-06-23:
Thank you for posting this. We receive a lot of open-source contributions, so not to call this official, here are some recommendations. 

1. **Follow NVIDIA's LeRobot Schema Guide**
The Isaac-GR00T repository provides detailed documentation on the LeRobot-compatible data schema[^4][^5]. You'll need to structure your data with:
    - Parquet files for episode chunks (e.g., `chunk-000.parquet`)
    - Metadata files in `meta/` directory:
        - `info.json`
        - `episodes.jsonl`
        - `tasks.jsonl`
    - A `modality.json` file for annotation specifications[^4]
    
2. **Use Third-Party Converters as Reference**
Projects like [ALOHA Data Converter](https://github.com/muhamuttaqien/ALOHADataConverter) demonstrate practical HDF5-to-LeRobot conversion[^2]. Key features include:

```bash
python convert_to_lerobot.py \
  --input_dir ./path/to/hdf5_data \
  --output_dir ./lerobot_dataset \
  --fps 30 \
  --task_string "Your_Task_Name" \
  --chunk_size 3
```

This script handles chunking, metadata generation, and Parquet conversion[^2].

3. **Leverage Isaac Lab Utilities**
Isaac Lab's `RobomimicDataCollector` provides a foundation for HDF5 data collection[^3]:

```python
from omni.isaac.lab_tasks.utils.data_collector import RobomimicDataCollector
collector = RobomimicDataCollector(task_name="Your_Task", log_dir="./logs")
# Add batched data during episodes
collector.flush_to_disk()  # Saves as HDF5
```

You'd need to extend this to output LeRobot's Parquet+metadata format.

_References_

[^1]: https://github.com/NVIDIA/Isaac-GR00T/issues

[^2]: https://github.com/muhamuttaqien/ALOH-Data-Converter

[^3]: https://isaac-sim.github.io/IsaacLab/source/api/lab_tasks/omni.isaac.lab_tasks.utils.data_collector.html

[^4]: https://github.com/NVIDIA/Isaac-GR00T/blob/main/getting_started/LeRobot_compatible_data_schema.md

[^5]: https://github.com/NVIDIA/Isaac-GR00T?tab=readme-ov-file

[^6]: https://github.com/NVIDIA/Isaac-GR00T/issues/222

[^7]: https://developer.nvidia.com/blog/fast-track-robot-learning-in-simulation-using-nvidia-isaac-lab/

[^8]: https://isaac-sim.github.io/IsaacLab/main/source/overview/environments.html

[^9]: https://isaac-sim.github.io/IsaacLab/main/source/api/lab_tasks/omni.isaac.lab_tasks.utils.data_collector.html

[^10]: https://docs.phospho.ai/learn/lerobot-dataset

Saketh-Vaishya, 2025-06-26:
thank you so much for sharing the knowledge @RandomOakForest

closed 2025-06-26 by devedgar

edisimo, 2025-07-10:
Will the HDF5-to-LeRobot data conversion pipeline be included in the upcoming Isaac Lab 2.2 release? @RandomOakForest