isaac-sim/IsaacLab #2741: [Bug Report] Semantic segmentation breaks for multiple envs

reeceomahoney, 2025-06-19:
### Describe the bug

When using a TiledCamera (or Camera) class with segmantic segmentation or instance segmentation, these break depending on how the camera is initialized. 

As per the note in https://github.com/isaac-sim/IsaacLab/blob/91ad4944f2b7fad29d52c04a5264a082bcaad71d/source/isaaclab/isaaclab/sensors/camera/camera.py#L151 isaac sim 4.5 breaks these outputs and so the instanceable flag is turned off. If the camera is defined in the scene cfg, the prim tree isn't fully constructed when this fix is called and so it doesn't work properly. As a result, the semantic labels only register for the first env, and all the others just register as unlabelled. 

This can however be fixed by moving the camera init into the post init

### Steps to reproduce

https://github.com/reeceomahoney/general_flow/blob/pcd/general_flow/envs/lift_env_cfg.py

The above env cfg shows a working example, where the camera config is in the post init. Moving this into the scene cfg breaks the semantic segmentation

### System Info

- Commit: [93167664]
- Isaac Sim Version: [4.5.0]
- OS: [Ubuntu 24.04]
- GPU: [RTX 2080 Ti]
- CUDA: [e.g. 12.4]
- GPU Driver: [550.144.03 ]

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

RandomOakForest, 2025-06-23:
Thank you for posting this. The team will review.

kellyguo11, 2025-07-08:
This should now be fixed in Isaac Sim 5.0, the open source repo is available at https://github.com/isaac-sim/IsaacSim

closed 2025-08-15 by reeceomahoney