isaac-sim/IsaacLab #2746: [Bug Report] IsaacSim & IsaacLab Mismatch

exaFLOPs26, 2025-06-20:
### Describe the bug

There is a mismatch with IsaacSim and IsaacLab.
As you can see IsaacSim and IsaacLab are showing different physics as I am pretty sure every setting is the same.

[IsaacSim]
https://github.com/user-attachments/assets/cabb12bd-8881-4a2f-90ab-997ce564aa6a

[IsaacLab]
https://github.com/user-attachments/assets/c68f5b96-c00c-4055-b1f5-fa0caf8a87f5

### Steps to reproduce

Below is my config for robot. I am giving stiffness 100 and damping 10 as in IsaacSim it looks fine. 

```
import isaaclab.sim as sim_utils
from isaaclab.actuators import ImplicitActuatorCfg
from isaaclab.assets.articulation import ArticulationCfg
##
# Configuration
##
ANUBIS_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path="/root/IsaacLab/source/isaaclab_assets/data/Robots/MM/anubis/anubis_omni.usd",
        activate_contact_sensors=False,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            rigid_body_enabled=True,
            kinematic_enabled=False,
            disable_gravity=False,
            max_linear_velocity=12.0,
            max_angular_velocity=12.0,
            max_depenetration_velocity=10.0,
            max_contact_impulse=10.0,
            stabilization_threshold=0.1,
        ),
        articulation_props=sim_utils.ArticulationRootPropertiesCfg(
            enabled_self_collisions=False,
            solver_position_iteration_count=16,
            solver_velocity_iteration_count=8,
            # fix_root_link=True,
        ),
    ), # --/renderer/shadercache/driverDiskCache/flush=true
    init_state=ArticulationCfg.InitialStateCfg(
        joint_pos={
            # base
            # "OmniR": 0.0,
            # "OmniFR": 0.0,
            # "OmniFL": 0.0,
            # arm <-> base (rad)
            "arm1_base_link_joint": 0.0,
            "arm2_base_link_joint": 0.0,
            # Right arm
            "link11_joint": -0.69289571,
            "link12_joint": 2.34048653,
            "link13_joint": -0.07679449,
            "link14_joint": 0.52359878,
            "link15_joint": -0.17453293,
            # Left arm
            "link21_joint": -0.69289571,
            "link22_joint": 2.34048653,
            "link23_joint": -0.07679449,
            "link24_joint": -0.52359878,
            "link25_joint": 0.17453293,
            # finger
            "gripper.*": 0.0,
        },
        joint_vel={".*": 0.0},
    ),
    actuators={
        "base": ImplicitActuatorCfg(
            joint_names_expr=["OmniR", "OmniFR", "OmniFL"],
            effort_limit_sim=1e10,
            velocity_limit_sim=1e10,
            stiffness=0.0,
            damping=1000000.0,  # tip:: For velocity control of the base with dummy mechanism, we recommend setting high damping gains to the joints. This ensures that the base remains unperturbed from external disturbances, such as an arm mounted on the base.
            friction=0.5,
        ),
        "dummy_spheres": ImplicitActuatorCfg(
            joint_names_expr=["OmniR_roller_.*", "OmniFR_roller_.*", "OmniFL_roller_.*"],
            effort_limit_sim=1e10,
            velocity_limit_sim=1e10,
            stiffness=0.0,
            damping=0.0001,
            friction=0.5,
        ),
        "arm_base": ImplicitActuatorCfg(
            joint_names_expr=["arm.*"],
            effort_limit_sim=1e10,
            velocity_limit_sim=1e10,
            stiffness= 100,
            damping= 10,
        ),
        "arm_link": ImplicitActuatorCfg(
            joint_names_expr=["link.*"],
            effort_limit_sim=1e10,
            velocity_limit_sim=1e10,
            stiffness= 100,
            damping= 10,
        ),
        "anubis_right_hand": ImplicitActuatorCfg(
            joint_names_expr=["gripper1.*"],
            effort_limit_sim=200.0,
            velocity_limit_sim=0.2,
            stiffness=2e4,
            damping=1e2,
            friction= 2.15,
        ),
        "anubis_left_hand": ImplicitActuatorCfg(
            joint_names_expr=["gripper2.*"],
            effort_limit_sim=200.0,
            velocity_limit_sim=0.2,
            stiffness=2e4,
            damping=1e2,
            friction= 2.15,
        ),
    },
)

```

### Checklist

- Gravity enabled for IsaacSim
- Checked that the below does not affect.
            max_linear_velocity=12.0,
            max_angular_velocity=12.0,
            max_depenetration_velocity=10.0,
            max_contact_impulse=10.0,
- Checked that when stiffness 1e5 and damping 1e2, then the arm stays still.
- No gravity compensation for Task-space IK? Also tried OSC but still not working.

### Additional context

My personal opinion is that action setting the target position code of Isaaclab is wrong. I will also go into details to help you all. Also adding gravity compensation for not only OSC but also for TSI like other simulation will be useful for users to teleop if you want Isaacsim or lab to be widely used for Robotic engineers!

RandomOakForest, 2025-06-23:
Thank you for posting this. Could you please share the versions of Isaac Sim and Isaac Lab you are using? Also could you elaborate on what exactly you mean by "showing different physics"?

exaFLOPs26, 2025-06-24:
@RandomOakForest  There is no gravity compensation for Differential IK action control right??