isaac-sim/IsaacLab #2747: [Question] When I reset the position of my robot it disappears from the simulation, what can be the reason? [closed the main one by mistake]

Abddo9, 2025-06-20:
### Question

I wanted to comment on [the original question](https://github.com/isaac-sim/IsaacLab/issues/2734), but I closed it by mistake and can not reopen it.

I am trying to use my own mobile robot with Isaac lab. I added the usd file to the data/robots folder and created the robot.py file.
I can control it OK now using set_joint_velocity_target, and set_joint_position_target.
However, whenever I try to reset it to its default position it just disappears from the simulation, without any errors, and the simulation keeps running, and I can see the robot in the stage tree, what can be the reason?

the robot.py code:

`
import isaaclab.sim as sim_utils
from isaaclab.actuators import ImplicitActuatorCfg
from isaaclab.assets import ArticulationCfg
from isaaclab_assets import ISAACLAB_ASSETS_DATA_DIR

Rangen_CFG = ArticulationCfg(
spawn=sim_utils.UsdFileCfg(
usd_path= ISAACLAB_ASSETS_DATA_DIR+"/robots/rangen/rangen.usd",
rigid_props=sim_utils.RigidBodyPropertiesCfg(
rigid_body_enabled=True,
max_linear_velocity=1000.0,
max_angular_velocity=1000.0,
max_depenetration_velocity=100.0,
enable_gyroscopic_forces=True,
),
articulation_props=sim_utils.ArticulationRootPropertiesCfg(
enabled_self_collisions=False,
solver_position_iteration_count=4,
solver_velocity_iteration_count=0,
sleep_threshold=0.005,
stabilization_threshold=0.001,
),
),
init_state=ArticulationCfg.InitialStateCfg(
pos=(0.0, 0.0, 0.0), joint_pos={".*": 0.0}
),
actuators={
"wheels_actuator": ImplicitActuatorCfg(
joint_names_expr=["fr_wheel", "fl_wheel", "rr_wheel", "rl_wheel"],
effort_limit=4000.0,
velocity_limit=1000.0,
stiffness=0.0,
damping=100.0,
),
"steering_wheels_actuator": ImplicitActuatorCfg(
joint_names_expr=["fr_steering_wheel", "fl_steering_wheel", "rr_steering_wheel", "rl_steering_wheel"],
effort_limit=4000.0, velocity_limit=1000.0, stiffness=100.0, damping=50.0
),
},
)
"""Configuration for a simple Rangen robot."""

@Sanjay1911 comment:

"Hi @Abddo9

This could be due to gravity sometimes and if you can still see your robot in stage tree, select the robot from your stage dropdown and press F. This would focus on your robot only from the stage and you should be able to see where it is.
Also what happens when you pause and replay the simulation?"

my new reply:

When I press F, it seems it is there but not shown.
I am trying it from inside an RL env [image]
Pause and unpause [here](https://github.com/user-attachments/assets/f3747e92-e7bb-4a2a-a3ec-5fa442ea6cf7)

Sanjay1911, 2025-06-21:
Hey, I'm not quite sure what exactly is happening just from the pictures. Were you able to open the Robot USD only in Isaac Sim ?
If yes, can you chck what happens when you press play? If you can also share your files and USD, I can maybe check and see if I can help you

Abddo9, 2025-06-23:
Thanks again @Sanjay1911, I managed to fix it.
For the benefit of others, the problem was that I was placing the robot very close to the ground plan. However, when I put it half a meter above the ground, it works fine.

And yes I was able to open the robot USD file, and play it, but not reset the position from the RL env.