isaac-sim/IsaacLab #2751: [Question] randomize_visual_texture_material.__init__() got an unexpected  keyword argument 'asset_cfg'

shiven2001, 2025-06-22:
Hello, I am using Isaac Sim 4.5.0.

I want to change the texture of one of my assets in my scene and I saw that I can use randomize_visual_texture_material for that. But I don't understand the implementation. The asset is as follows:
`
- {ENV_REGEX_NS}
    - environment
        - wall (xform with rigidbody)
            - wall (plane that I want to apply texture)
                - visuals (folder)
                    -  wall (material for default use for wall plane which uses a .png file)

I want that on reset, the png file changes for the material (I have 50 png files and I want it to be random) 

I think randomize_visual_texture_material can do this but I tried adding this to my interactive scene:

      # wall
      wall = RigidObjectCfg(
          prim_path="{ENV_REGEX_NS}/environment/wall",
          spawn=None,
          init_state=RigidObjectCfg.InitialStateCfg(),
      )

and my event cfg as:

      # Event cfg
      @configclass
      class EventCfg:
          """Configuration for events with fixed orientation."""
      
          reset_scene = EventTerm(func=mdp.reset_scene_to_default, mode="reset")
      
          change_plane_texture = EventTerm(
              func=mdp.randomize_visual_texture_material,
              mode="reset",
              params={
                  "asset_cfg": SceneEntityCfg("wall"),
                  "texture_paths": TEXTURE_FILES,
                  "event_name": "randomize_wall_texture",
                  "texture_rotation": (0.0, 0.0),  # No rotation (in radians)
              },
          )

where
      
      # Texture directory
      TEXTURE_DIR = os.path.join(script_dir, "usds/synthetic/")
      # Get all PNG files
      TEXTURE_FILES = glob.glob(os.path.join(TEXTURE_DIR, "*.png"))
      sys.path.append(script_dir)
       

but it gives error:
TypeError: randomize_visual_texture_material.__init__() got an unexpected 
keyword argument 'asset_cfg'

RandomOakForest, 2025-06-23:
Thanks for posting this. A similar issue was reported in #2388. Would the workarounds in the comments work in your case?

shiven2001, 2025-06-24:
Hi, I got it working. Thanks, since I didn't see an implementation of it in IsaacLab, I will share my code here for future reference for others:
      
      import torch
      from isaaclab.envs import ManagerBasedEnv
      from isaaclab.managers import SceneEntityCfg as SceneEntityCfg
      import os
      import sys
      import logging
      import numpy as np
      import random
      import omni
      import isaaclab.sim as sim_utils
      from isaaclab.assets import RigidObject, RigidObjectCollection
      import omni.usd
      from isaacsim.core.utils.extensions import enable_extension
      from pxr import Gf, Sdf, UsdGeom, Vt
      import carb
      from pxr import UsdShade, Sdf
      
      # Set up logging
      logging.basicConfig(level=logging.INFO)
      logger = logging.getLogger(__name__)
      
      script_dir = os.path.dirname(os.path.abspath(__file__))
      sys.path.append(script_dir)
      import math
      
      
      def randomize_texture(
          env: ManagerBasedEnv,
          env_ids: torch.Tensor,
          asset_cfg: SceneEntityCfg,
          event_name: str,
          texture_paths: list[str],
          texture_rotation: tuple[float, float] = (0.0, 0.0),
      ):
          asset = env.scene[asset_cfg.name]
          stage = env.scene.stage
      
          for env_id in env_ids:
              prim_path = f"{env.scene.env_ns}/env_{env_id}/environment/{asset_cfg.name}"
              mtl_prim_path = f"{env.scene.env_ns}/env_{env_id}/environment/{asset_cfg.name}/{asset_cfg.name}/visuals/{asset_cfg.name}_material"
              prim = stage.GetPrimAtPath(prim_path)
              mtl_prim = stage.GetPrimAtPath(mtl_prim_path)
      
              # Randomly select a texture from the provided list
              selected_texture = random.choice(texture_paths)
      
              omni.usd.create_material_input(
                  mtl_prim,
                  "diffuse_texture",
                  selected_texture,
                  Sdf.ValueTypeNames.Asset,
              )
      

I am very curious though, is the replicator not compatible with IsaacLab? I even made a custom function based on randomize_visual_texture_material which doesn't have the  TypeError: randomize_visual_texture_material.init() got an unexpected
keyword argument 'asset_cfg' problem and runs without any errors, it still didn't do anything when I called functions like

rep.randomizer.texture( ) or rep.randomizer.color( )... 

It would be insanely helpful if replicator was compatible with isaaclab... is my implementation wrong or something? I had this:

      import torch
      from isaaclab.envs import ManagerBasedEnv
      from isaaclab.managers import SceneEntityCfg as SceneEntityCfg
      import os
      import sys
      import logging
      import numpy as np
[... 60 lines cut ...]
      
              carb.log_info(f"Matched prims for {pattern_with_visuals}: {matching_prims}")
              if matching_prims:
                  prim_path = pattern_with_visuals
              else:
                  prim_path = f"{asset_main_prim_path}/.*"
                  carb.log_info(
                      f"Pattern '{pattern_with_visuals}' found no prims. Falling back to '{prim_path}' for texture randomization."
                  )
      
              # Create the omni-graph node for the randomization term
              def rep_texture_randomization():
                  prims_group = rep.get.prims(path_pattern=prim_path)
      
                  with prims_group:
                      rep.randomizer.texture(
                          textures=texture_paths,
                          project_uvw=True,
                          texture_rotate=rep.distribution.uniform(*texture_rotation),
                      )
                  return prims_group.node
      
              # Register the event to the replicator
              with rep.trigger.on_custom_event(event_name=event_name):
                  rep_texture_randomization()