isaac-sim/IsaacLab #2756: [Bug Report] torch.distributed.run fails with CUDA_VISIBLE_DEVICES due to "CUDA being in bad state" in Isaac Sim

Revindell, 2025-06-23:
Hi Isaac Lab Team,

I'm encountering a critical error when trying to run a distributed training script using torch.distributed.run with CUDA_VISIBLE_DEVICES set to a subset of available GPUs. It appears that although the environment variable is set, Isaac Sim fails to correctly initialize the specified GPUs, leading to a "CUDA being in bad state" error and preventing any device from being created.

This issue prevents multi-GPU distributed training on a specific selection of GPUs.

# Steps to Reproduce:

## System Configuration:

OS: Ubuntu 22.04.5 LTS
GPU: 8 x NVIDIA H100 80GB HBM3
NVIDIA Driver Version: 550.163.01
Isaac Sim Version: 4.5 (based on log paths)
Environment: micromamba
Command to run:
The following command is executed from the root of the Isaac Lab workspace. It attempts to launch a 2-process distributed training job on GPUs with physical IDs 5 and 6.

```Bash

CUDA_VISIBLE_DEVICES=5,6 python -m torch.distributed.run --nnodes=1 --nproc_per_node=2 scripts/reinforcement_learning/skrl/train.
py --task=Isaac-Cartpole-v0 --headless --distributed
```

##Expected Behavior:

The script should launch three processes, with each process being assigned to one of the specified GPUs (5, 6,). Isaac Sim should initialize correctly on each process, mapping cuda:0 for each process to the corresponding physical GPU (e.g., process 0 -> physical GPU 5, process 1 -> physical GPU 6, etc.).

##Actual Behavior:

The script fails to initialize the simulation environment. The logs indicate that although torch.distributed seems to start correctly, the Isaac Sim/Omniverse application foundation cannot create a GPU device.

##Key error messages from the log include:

A warning about CUDA_VISIBLE_DEVICES being set:

2025-06-23 13:59:03 [16,707ms] [Warning] [carb.cudainterop.plugin] CUDA_VISIBLE_DEVICES environment variable is set.
2025-06-23 13:59:03 [16,707ms] [Warning] [carb.cudainterop.plugin] Note CUDA device enumeration and Omniverse device enumeration are different.
2025-06-23 13:59:03 [16,707ms] [Warning] [carb.cudainterop.plugin] Setting CUDA_VISIBLE_DEVICES can lead to undesired behavior or crashes.
A repeated warning that the GPUs are being skipped because CUDA is in a "bad state":

2025-06-23 13:59:04 [16,983ms] [Warning] [gpu.foundation.plugin] Skipping NVIDIA GPU due CUDA being in bad state: NVIDIA H100 80GB HBM3
2025-06-23 13:59:04 [16,983ms] [Warning] [gpu.foundation.plugin] Please restart your system if CUDA is known to work in your system.
A final critical error stating no device could be created:

2025-06-23 13:59:04 [17,106ms] [Error] [gpu.foundation.plugin] No device could be created.
This suggests a conflict between how torch.distributed and Isaac Sim's gpu.foundation.plugin interpret the CUDA_VISIBLE_DEVICES variable, ultimately leading to a failure in the GPU initialization process within the simulation.

## Full Print Out
$ CUDA_VISIBLE_DEVICES=5,6 python -m torch.distributed.run --nnodes=1 --nproc_per_node=2 scripts/reinforcement_learning/skrl/train.
py --task=Isaac-Cartpole-v0 --headless --distributed
WARNING:__main__:
*****************************************
Setting OMP_NUM_THREADS environment variable for each process to be 1 in default, to avoid your system being overloaded, please further tune the variable for optimal performance in your application as needed. 
*****************************************
[INFO][AppLauncher]: Using device: cuda:0[INFO][AppLauncher]: Using device: cuda:0

[INFO][AppLauncher]: Loading experience file: /raid/ws_df/ws/IsaacLab/apps/isaaclab.python.headless.kit
[INFO][AppLauncher]: Loading experience file: /raid/ws_df/ws/IsaacLab/apps/isaaclab.python.headless.kit
[Warning] [simulation_app.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [simulation_app.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [simulation_app.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
[Warning] [simulation_app.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/home/deyu.fu/micromamba/envs/dex/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.5/user.config.json'
[Info] [carb] Logging to file: /home/deyu.fu/micromamba/envs/dex/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.5/kit_20250623_141446.log
Loading user config located at: '/home/deyu.fu/micromamba/envs/dex/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.5/user.config.json'
2025-06-23 14:14:46 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /home/deyu.fu/micromamba/envs/dex/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.5/kit_20250623_141446.log
2025-06-23 14:14:46 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2025-06-23 14:14:46 [2ms] [Warning] [omni.ext.plugin] [ext: rendering_modes] Extensions config 'extension.toml' doesn't exist '/raid/ws_df/ws/IsaacLab/apps/rendering_modes' or '/raid/ws_df/ws/IsaacLab/apps/rendering_modes/config'
[... 650 lines cut ...]
| 6   | NVIDIA H100 80GB HBM3            |        |     | 81559   MB | 10de      | 0          |
|     |                                  |        |     |            | 2330      | fa8479ed.. |
|     |                                  |        |     |            | d1        |            |
|---------------------------------------------------------------------------------------------|
| 7   | NVIDIA H100 80GB HBM3            |        |     | 81559   MB | 10de      | 0          |
|     |                                  |        |     |            | 2330      | b58259d5.. |
|     |                                  |        |     |            | df        |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 5.15.0-1078-nvidia
| Processor: Intel(R) Xeon(R) Platinum 8480C
| Cores: 112 | Logical Cores: 224
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 2063936 | Free Memory: 1991726
| Total Page/Swap (MB): 0 | Free Page/Swap: 0
|---------------------------------------------------------------------------------------------|
2025-06-23 14:16:07 [80,661ms] [Error] [gpu.foundation.plugin] No device could be created. Some known system issues:
- The driver is not installed properly and requires a clean re-install.
- Your GPUs do not support RayTracing: DXR or Vulkan ray_tracing, or hardware is excluded due to performance.
- The driver cannot enumerate any GPU: driver, display, TCC mode or a docker issue. For Vulkan, test it with Vulkaninfo tool from Vulkan SDK, instead of nvidia-smi.
- For Ubuntu, it requires server-xorg-core 1.20.7+ and a display to work without --no-window.
- For Linux dockers, the setup is not complete. Install the latest driver, xServer and NVIDIA container runtime.

2025-06-23 14:16:07 [81,301ms] [Error] [gpu.foundation.plugin] Invalid getDeviceInfo parameters.
2025-06-23 14:16:07 [81,301ms] [Warning] [omni.physx.foundation.plugin] PhysXFoundation: Couldn't get driver version, failed to getDeviceInfo
2025-06-23 14:16:07 [81,301ms] [Warning] [omni.physx.foundation.plugin] PhysXFoundation: Unable to get IGpuFoundation, GpuDevices or Graphics!

RandomOakForest, 2025-06-23:
Thank you for posting this. The team will review.

AntoineRichard, 2026-08-27:
@deyuf thanks for the sharp diagnosis, and sorry for the delay. The renderer was selected through `/renderer/activeGpu`, which indexes the graphics device list and isn't filtered by `CUDA_VISIBLE_DEVICES`; it now uses `/renderer/multiGpu/activeCudaGpus`, a CUDA device index, so masked subsets like `5,6` resolve correctly. That shipped in https://github.com/isaac-sim/IsaacLab/pull/7057 (`isaaclab` package v16.2.2).

Please upgrade, and reopen with a fresh log if it persists. Unrelated: `docs/source/features/multi_gpu.rst` covers the two-rank NCCL P2P hang and `NCCL_P2P_DISABLE=1`.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard